lexus2k/lcdgfx #2: Specify offset/size for menu

alastaira, 2019-12-12:
Thankyou for this library - it seems robust and very well-documented. I particularly like the built-in scrolling menu capability.
However, I'm using a (relatively common, I believe) 128x64 "yellow/blue" OLED display based on the SSD1306. Essentially, it's a monochrome display, but the top 16 rows are yellow, while the bottom 48 pixels are blue (like this http://www.embeddedadventures.com/oled_display_128x64_OLED-12864-YB.html) 
It looks very ugly when the menu changes colour halfway down the screen, so I'd like to specify that the menu should only occupy the bottom 48 pixels - leaving the top 16 rows free to draw a separate topbar on the canvas. 

I imagine this could be achieved relatively easily as a function overload to the showMenu() and drawMenuItem functions, specifying additional optional parameters for desired x, y, width and height? They seem to currently have hardcoded values.

lexus2k, 2019-12-13:
Hi,

This is good idea. I will work under implementation

lexus2k, 2019-12-13:
Hi,

could you please check v1.0.0 branch and let me know if first changes fit your needs?

`display.createMenu( &menu, menuItems, sizeof(menuItems) / sizeof(char *), (NanoRect){{0,16},{0,0}} );`

alastaira, 2019-12-15:
The offset works great, thankyou, and adding the {0,16} parameter makes the menu look much better on this display - centred in the screen and avoiding the yellow/blue interface.

I got a bit confused by the behaviour of the second parameter though - the width/height. It seems that supplying {0,0} means "make the menu centred on the screen, based on the supplied top/left corner", which is fine, but supplying other non-zero values seems to not quite lead to the menu size I expect. For example, to create a menu high enough to accomodate a single line of text using the 6x8 font, I seem to need to supply a height of {0,40}. To accomodate two lines of text, it's {0,48}.
Supplying a value of {0,36}, say, creates an apparently single line height box, but doesn't write any text menu items. Is this to do with the hardcoded values that are used to manually pad the menus (I still see various -4, -5 etc. values in the menu code?)

lexus2k, 2019-12-15:
Hi @alastaira,

Well, NanoRect doesn't contain width and height fields really. According to documentation NanoRect is the structure consisting of two fields: NanoPoint p1, NanoPoint p2. p1 is top-left point of rectangle area, p2 is right-bottom. The width is calculated as `p2.x - p1.x + 1`. And other moment is that `createMenu` method allows to specify zero-values for p2 field. In that case, the library will automatically set p2 to right-bottom point of display screen.

closed 2019-12-19 by lexus2k