lexus2k/lcdgfx #3: Attiny84

alxtr1bal, 2019-12-21:
Hello, i'm trying to make this work with an attiny84 , but i'm not able to. (with SSD1306).
The oled is working fine on a atmega328p 
Working also with another library i found (ssd1306_minimal), but i wanted to try yours.

Tried DisplaySSD1306_128x64_I2C display(-1, { -1, 0x3C, 4, 6, 0 });
or DisplaySSD1306_128x64_I2C display(-1);

But nothing happens... i'd be glad if you could help;

Alexis

lexus2k, 2019-12-22:
Hi @alxtr1bal,

I checked the code again, compared to pinout of Attiny84 [Attiny84 pinout](https://camo.githubusercontent.com/d46f3f004aaf977040d933ae5eaf25d22d33eac1/687474703a2f2f6472617a7a792e636f6d2f652f696d672f50696e6f7574543834612e6a7067), and didn't find any issues.
Do you have any logic analyzer or oscilloscope? 
Can you run example at 8MHz?

I don't have Attiny84 by hand right now, but I will let you know once I get it.

lexus2k, 2019-12-22:
Hi @alxtr1bal 

I found that new versions of AVR compilers use high code optimization, and software support is very dependent on generated asm code. So, I switch software implementation to built-in avr delay functions. Can you check them?
It is [v1.0.0_dev](https://github.com/lexus2k/lcdgfx/tree/v1.0.0_dev) branch

lexus2k, 2019-12-26:
Did you have a chance to verify the fix?

alxtr1bal, 2019-12-27:
Hey, i'm sorry for the delay ! After a lot of testing i found out that i had to put a little delay before starting the screen, otherwise the screen was never showing up ! So nothing to do with your code , i'm sorry if i had let you work for that :(

alxtr1bal, 2019-12-27:
I will have a look at your dev branch.

closed 2019-12-27 by alxtr1bal

lexus2k, 2020-01-01:
Let me know if it works. I will propagate all changes to master branch