lexus2k/lcdgfx #4: ssd1306_demo not working on ESP32

srxdevnet, 2020-01-02:
Hello there!
I´m trying to use the demo with ESP32 (DOIT devkit) but nothing shows up on the oled screen. If I use your C library (SSD1306) with the same circuit, works like charm.
Any ideas to test?

Thank you in advance!

lexus2k, 2020-01-02:
Hi @srxdevnet 

Could you please describe the issue in details? You described above 2 use cases, so I have questions on them:

1. First case: ESP32 (DOIT devkit)

a) Do you use ssd1306_demo for testing? 
b) What is DOIT devkit? If it has official name, it will be very helpful if you provided it.
c) What SDK do you use in first case: Arduino IDE or ESP32-IDF? What is the version 4.2?

2. Second case (C-library)

d) How do you use C-library? Do you have code example?
e) Do you use ESP-IDF for the second case?

Best regards, Alexey

lexus2k, 2020-01-02:
I verified command line compilation using ESP32-IDF 3.3.0 stable release and ESP32 dev module (i2c ssd1306), and it works.
Next, I verified ssd1306_demo from Arduino IDE using official esp32 package, and it works.
Also, you can try to put small delay before initializing display. Maybe, your esp32 powers on too quickly, and display hardware is not ready yet.

lexus2k, 2020-01-02:
Hi again.

I fixed the problem (see commit above). Could you please check [v1.0.0_dev](https://github.com/lexus2k/lcdgfx/tree/v1.0.0_dev) branch and let me know if it works

Thanks

srxdevnet, 2020-01-02:
Hi! Thanks for your fast answer!
I´ll clarify the cases for you.
The hardware i´m using is this:

ESP32
https://articulo.mercadolibre.com.ar/MLA-791810372-placa-nodemcu32-wifi-bluetooth-esp32-wroom-arduino-nubbeo-_JM?quantity=1 

SSD1306:
https://articulo.mercadolibre.com.ar/MLA-705923840-display-oled-13-azul-128x64-i2c-sh1106-arduino-nubbeo-_JM?quantity=1

The circuit wiring is the same as specified for the demo in both libs (SSD3106 & LCDGFX)

I´m using VSCode + PlatformIO to compile and flash the ESP32

In the first case i´m compling ssd1306_demo of the SSD1306 library (C)

In the second case i´m compiling ssd1306_demo of the LCDGFX library (C++). In this case isn´t working.

I´ll try the v1.0.0_dev branch if it works.

Thanks in advance!

srxdevnet, 2020-01-02:
Hi There!

Well, I´ve tested v1.0.0_dev branch and works great in Arduino IDE.
In PlatformIO I´ll test and give you feedback if you like.

Thank you very much!

closed 2020-01-02 by srxdevnet