lexus2k/lcdgfx #5: SSD1327 128x128 ESP32

Mael19, 2020-02-14:
Hello, I can't seem to make my SSD1327 (i2c) display work with my ESP32 board, either with this or the u8g2 library.

I'm using this line to set the display up :
DisplaySSD1327_128x128_I2C display(-1);  

Does anyone has an idea ? I'm starting to think of a hardware failure...

lexus2k, 2020-02-15:
Hello,

Can you try to add pullup resistors for scl and sda lines.
I noticed, that you asked the same question in u8g2 issues. According to pictures from there, maybe you have issues with i2c , or power supply for your setup.
Do you have hardware logic analyzer?

Mael19, 2020-02-15:
I tried the pullup resistors, but nothing seem to change. And no I don't have any logic analyser, in fact I got very few electronic material.

lexus2k, 2020-02-17:
If you have logic analyzer, it would be very nice. Since, at this point, I don't have any inputs about what's happening on communication lines for your display.
At least this one would be enough: [Logic Analyzer](https://aliexpress.ru/item/4000671169342.html?spm=a2g0o.productlist.0.0.22d98300lhGSzG&algo_pvid=a0118b21-65e8-41a8-b1fd-e64cde70f40e&algo_expid=a0118b21-65e8-41a8-b1fd-e64cde70f40e-4&btsid=0b8b158f15819219132362182eb4f9&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)

Mael19, 2020-02-17:
Well at the moment I don't have much time to spend on this, maybe I'll brought it to a friend who is better at electronics than me or order an other screen...

lexus2k, 2020-07-02:
I'm closing the issue, since there are no any inputs.

closed 2020-07-02 by lexus2k