lexus2k/lcdgfx #6: SSD1327 128x128 I2C some fonts crippled

wwebers, 2020-02-16:
**Describe the bug**
I use a Waveshare 128x128 monochrome OLED connected (and jumpered) via I2C at a Arduino Nano (old bootloader). The display works fine when explicitely using the display I2C address (0x3D in my case). However, some fonts displayed crippled when using the NanoEngine library.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the 1.5inch OLED to A4 and A5 at the Nano board
2. Open menu demo sketch from the library
3. Choose NanoEngine1 as engine
4. Select e.g. "free_calibri11x12" font

**Expected behavior**
I would expect that every font will be displayed nicely.

**Screenshots**
[image]
[image]

**Please complete the following information:**
 - library version: 1.0.2
 - LCD display type: Waveshare 1.5inch monochrome OLED
 - Platform: Atmega328p
 - IDE: PlatformIO and Arduino 1.8.11

**Additional context**
Sometimes, but not always, this works fine when using NanoEngine4 or NanoEngine8:

[image]

lexus2k, 2020-02-17:
Hello,

If  I understand correctly from your description, you're using NanoEngine1 along with 4-bit grey-scale display. This functionality is not implemented right now. You should use NanoEngine with the same bitness as your display:
SSD1327 is 4-bit display, then please, use NanoEngine4.

Next, the font, you pointed above, is FreeType font, not fixed one. So, you need to use setFreeFont method instead of setFixedFont.

Here is working example:
```.cpp
#include "lcdgfx.h"

DisplaySSD1327_128x128_I2C display(-1);

const PROGMEM uint8_t heartImage[8] =
{
    0B00001110,
    0B00011111,
    0B00111111,
    0B01111110,
    0B01111110,
    0B00111101,
    0B00011001,
    0B00001110
};

typedef NanoEngine4<DisplaySSD1327_128x128_I2C> MyEngine;

MyEngine engine(display);
NanoTextMenuItem<MyEngine::TilerT> item1("First menu item");
NanoTestMenuItem<MyEngine::TilerT> item2;
NanoTextMenuItem<MyEngine::TilerT> item3("Demo");
NanoTestMenuItem<MyEngine::TilerT> item4;
NanoFixedWidthMenu<MyEngine::TilerT> menu( {0,0}, {128,64} );

uint16_t lastMillis;

void setup()
{
    display.begin();
    display.setFreeFont(free_calibri11x12);

    engine.setFrameRate( 30 );
    engine.begin();

    engine.getCanvas().setFreeFont(free_calibri11x12);
    engine.getCanvas().setMode(CANVAS_MODE_TRANSPARENT);

    menu.add( item1 );
    menu.add( item2 );
    menu.add( item3 );
    menu.add( item4 );
    engine.insert( menu );

    engine.refresh();
    lastMillis = millis();
}

void loop()
{
    if (static_cast<uint16_t>(millis() - lastMillis) > 2000 )
    {
        lastMillis = millis();
        menu.down();
    }
    if (!engine.nextFrame()) return;
    engine.update();
    engine.display();
}
```

wwebers, 2020-02-18:
Thanks, didn't know the SS1327 was 4-bit. 
I tested setFreeFont() and it works fine now. Just to figure out when to use setFixedFont() and when to use setFreeFont()...

closed 2020-02-18 by wwebers