lexus2k/lcdgfx #7: setOffset does not seem to work?

Didel, 2020-03-02:
Hi,

I am working on a project that requires some LCD interaction, so after searching for alternatives for the Adafruit LCD libraries I stumbled upon LCDGFX. After some research, I got it to work in my setup: An NodeMCU V3 ESP8266 with an 128x60 ST7735 Lcd screen. It is connected with an SPI connection.

After adjusting the code it seems to work fine, displaying everything I want in the correct colors. Running the Snowflake demo is fun! Except... The screen area falling off the screen with 1px or 2px both on X and Y axis. I've seen this happening too with other libraries, but they often supply an offset function to fix this which has been working fine with other libraries.

I see that LCDGFX also has an [setOffset()](https://lexus2k.github.io/lcdgfx/class_nano_display_base.html#a182078088dd5caabbd242d75a7bbe48a) method, but calling `display.setOffset(1, 2);` doesn't seem to have have any effect. This is about as less code as I could create:

```cpp
#include "lcdgfx.h"

#define TFT_CS         D3
#define TFT_RST        D4                                            
#define TFT_DC         D2

DisplayST7735_128x160x16_SPI display(TFT_RST,{-1, TFT_CS, TFT_DC, 0,-1,-1});

void setup() {
  display.begin();
  display.setOffset(1, 2);
  display.clear();
  display.fill(0x000);
}

void loop() {
  
}
```

Which results in this:

[image]

As you can see, an offset of 1px and 2px (guesstimate) would fix the random pixels near the edge. However, no matter what values I supply to `setOffset()`, I get the same result. Can you help me out? Am I doing something wrong?

lexus2k, 2020-03-03:
Hello,

Thank you for reporting the issue.
What if you set 0,0 offset? And other question, can you draw 2 rectangles (0,0)-(127,159) and another one with (2,2)-(125,157) using offset (0,0)?

From your screenshot, it looks like offset should be (2,0). Can you check?

Best regards

Didel, 2020-03-03:
Hi @lexus2k, 

Thank you for providing this great library!

I quickly created the following code to reproduce your request:

```c++
#include "lcdgfx.h"

#define TFT_CS         D3
#define TFT_RST        D4
#define TFT_DC         D2

DisplayST7735_128x160x16_SPI display(TFT_RST,{-1, TFT_CS, TFT_DC, 0,-1,-1});

void setup() {
  display.begin();
  display.setOffset(0, 0);
  display.clear();

  // Red
  display.setColor(RGB_COLOR16(255, 0, 0));
  display.drawRect(0, 0, 127, 159);

  // Green
  display.setColor(RGB_COLOR16(0, 255, 0));
  display.drawRect(2, 2, 125, 157);
}

void loop() { }
```

Which gave the following result:

[image]

The strange thing is: Even when I change the offset line to something like `display.setOffset(25, 25);`, and after I re-compiled and re-uploaded to the ESP8266, that gives the exact same visual result (while I would expect that to provide a clearly shifted result). I even changed one of the colors to something different to be sure that I got the newest code on the ESP. The color change was visible, but no offset or whatsoever.

Thanks again for helping out!

lexus2k, 2020-03-04:
I'm so sorry, the setOffset method doesn't do anything useful now.
I will provide the fix soon.
Can you check the same example with other library: ssd1306 (just for test), it has valid setOffset function for ili9341 displays.

lexus2k, 2020-03-04:
Also, I committed the fix to dev branch (you can see it above) for lcdgfx library. Could you please, check it and let me know if it works in all modes?

Didel, 2020-03-04:
Hi @lexus2k, 

Thanks a lot for your quick response!

I did some testing with the 1.0.0-dev branch of the library, and it appears to work exactly as intended! I've ran some of the supplied examples (including the ST7735 demo), included the appropriate offset (which turned out to be 2, 1 in this case) and that worked like a charm!

I did notice one thing though, but I'm not sure if that is specific to this issue or something common for all supported screens: When you set an offset _and_ also set a rotation (with `display.getInterface().setRotation();`), the offset is not 'adjusted' for the rotation. So, when you `setOffset(2, 1)` and also `setRotation(1)`, the offset is calculated for the _rotated_ coordinates, which I think is incorrect. It is easily fixed by changing `setOffset(x, y)` to `setOffset(y, x)`, but I don't think that should be necessary as it makes it harder to use screen-rotation during run-time (as seen in e.g. the demo). When rotated, the X and Y coordinates are switched (X becomes Y and vice versa), but the offset should not change, as the offset is not a property of the contents rendered on the screen but instead is a property for the specific piece of hardware. The result is that when you rotate the screen by 90 or 270 degrees, you still have to adjust the offset to be able to use all the pixels.

lexus2k, 2020-03-05:
Ok, thank you for detailed report.
I will prepare final fix as soon, as I have enough time. I'm glad that the fix above helped you.

lexus2k, 2020-03-07:
I've updated setOffset() and setRotation() functions, the changes should fix the problem you're described above

Didel, 2020-03-07:
Hi @lexus2k,

Thanks for your work! It appears that that problem is now fixed as well! Thanks a lot for your quick responses, and again thank you very much for making LCDGFX in the first place!

lexus2k, 2020-03-08:
Thank you. Let me know if find more issues.

closed 2020-03-08 by lexus2k