lexus2k/lcdgfx #8: Set font character spacing

octave, 2020-03-12:
Hi,

First of all thank you for this great library! I was wondering if there is a way to set the font character spacing when using `setFreeFont` or `setFixedFont` ?

Thank you!

lexus2k, 2020-03-13:
Hello,

Unfortunately, there is no such method for now. But I've learned that it is easy to add method `setSpacing()` to `NanoFont` class and `setFontSpacing()` to display class to have something like this:

```.cpp
    display.setFreeFont( free_calibri11x12, free_calibri11x12_cyrillic );
    display.setFontSpacing(2);
    display.printFixed(0,  8, u8"This is Русский", STYLE_NORMAL);
```
What do you think?

Best regards.

octave, 2020-03-13:
That would be awesome!

lexus2k, 2020-03-13:
Please, find my commit above.
Let me know if it works for you as expected.

```.cpp
    display.setFreeFont(...)
    display.setFontSpacing(2);
    // or
    display.getFont().setSpacing(2);
```

octave, 2020-03-13:
It works!! Thank you that was super fast!

closed 2020-03-14 by lexus2k