lexus2k/lcdgfx #12: SSD1331 support with HSPI on AI THINKER ESP32 CAM

jefflcc, 2020-03-30:
Hi,
Is HSPI supported? How do I set it is supported?
I wasn't able to get the HSPI on the AI Thinker esp32-cam board to work with the OLED SSD1331
with following setting. (not using the SDcard slot)
      DisplaySSD1331_96x64x8_SPI display(0,{0, 3, 15, 0,-1,-1});  rst =0, busID=0, cs=3,dc=15 
The VSPI on ESP32 is working fine with the OLED SSD1331 with the following setting. 
      DisplaySSD1331_96x64x8_SPI display(32,{-1, 33, 5, 0,-1,-1}); rst =32, busID=-1, cs=33,dc=5

Thanks in advance

lexus2k, 2020-03-30:
Hello,

So, if you leave DC pin and in your VSPI example (you are allowed to use any pin you want) and want to use HSPI pins, then your hardware configuration should be like this:
1. DC = 33
2. RST = 32
3. CLK = 14
4. MOSI = 13
5. CS = 15

`DisplaySSD1331_96x64x8_SPI display(32,{0, 15, 33, 0, 14, 13});`

By default if you don't specify last 2 parameters (scl:-1, sda:-1), the library uses 14 and 13 numbers.
It seems, that you have swapped 2 parameters in SPlatformSpiConfig structure: dc and cs.

jefflcc, 2020-03-31:
Hi, 

Thanks for the advice,

I was wondering if HSPI on AI thinker esp32cam to drive ssd1331 may be an issue somewhere.
based on my below findings.

I swapped the cs and dc just in case.
Both below setting also didn't work
DisplaySSD1331_96x64x8_SPI display(0,{0, 3, 15, 0, 14, 13}); rst =0, busID=0, cs=15,dc=3
DisplaySSD1331_96x64x8_SPI display(0,{0, 3, 15, 0,-1,-1}); rst =0, busID=0, cs=15,dc=3
The same HW and pins setup above,using another display spi driver is able to drive a 3.5" LCD without any issue. 

On the other ESP32 dev HW using VSPI pins, I've no issue driving SSD1331 whether the dc/cs pins are swapped as long as long as they go to the corresponding pins.

lexus2k, 2020-04-01:
Are you using Arduino IDE?
Arduino interface by default is attached to standard SPI object, which maps to VSPI on ESP32. I added ArduinoSpi2 interface to the library, which is attached to HSPI.

Please, download latest library sources from 1.0.0_dev branch and use the following initialization as example (you can set the pins you need):

```.cpp
// rst, dc, cs, dc (again!), clk, mosi, freq
DisplaySSD1331_96x64x8_CustomSPI<ArduinoSpi2> display(21, 22, 15, 22, -1, -1, 8000000);
```

jefflcc, 2020-04-01:
Hi,
yes, I'm using Arduino IDE.
This works perfectly now. 
But it caused a compile error when I target for esp8266, 
so have to comment out #define CONFIG_ARDUINO_SPI2_AVAILABLE	
in io.h file when using esp8266. Also regress with esp32 dev/nano without issue.

Thanks again:)

lexus2k, 2020-04-02:
Got it. Thank you very much for testing.
I've integrated fix for esp8266

jefflcc, 2020-04-02:
Awesome.

commit caf05be9 (2020-04-01): Esp32 HSPI for Arduino interface fix #12
closed 2020-04-02 by jefflcc