lexus2k/lcdgfx #13: FixedFont unocode\cyrillic

Voha888, 2020-04-03:
**FIXED FONT** is more convenient because when another inscription is displayed at the same place, the old text disappears. But I did not find an example of how to generate a unicode font for use with the **display.setFixedFont** function? Perhaps there is a ready-made cyrilic font, or an example of how to make such a font?

lexus2k, 2020-04-04:
Hello,

The solution depends on how you're going to use non-ascii chars, what type of editor you use for your project, and if your source files are in utf-8 encoding by default or not.
Anyway, download latest library version from v1.0.0_dev branch.
Then you can generate required fonts using several ways:

1. Using GLCD Font Creator program

2. Using fontgenerator python script to generate Fixed font format from ttf font you like:
> ./fontgenerator.py --ttf consola.ttf -s 8 -g 0 126 -f old -d -t Hello > eng_font.cpp
> // for ascii layout
> ./fontgenerator.py --ttf consola.ttf -s 8 -g 191 0 -g А 63 -f old -d -t Привет > rus_font.cpp
> // or for utf-8 layout
> ./fontgenerator.py --ttf consola.ttf -s 8 -g А 63 -f old -d -t Привет > rus_font.cpp

Then you need to make sure that the width of both generated fonts is the same, so you can use them together:
```.cpp
    display.setFixedFont(your_eng_font);
    display.getFont().loadSecondaryFont(your_rus_font);
    display.write("Привет");
```

3. Using fontgenerator python script to generate Free font of fixed width from ttf font you like:
> // utf-8 only
> ./fontgenerator.py --ttf consola.ttf -s 8 -fw -g 0 127 -g А 63 -f new -d -t Привет > eng_rus_output.cpp

```.cpp
    display.setFreeFont(your_free_font);
    display.write("Привет");
```

Best regards

Voha888, 2020-04-04:
Thank You very match! I have a some issue with fontgenerator, when I trying to generate a font on python 3.x it gives errors (it seems syntactic), but on python 2.x (which is recommended) it says they can't import a module from modules (somehow). I will try on another PC, or from under linux (judging by your examples, you use it). Thanks again, now at least I know how it will work. I’ll attach finished fonts here

Алексей, огромное спасибо. Попробую разобраться. Правда у меня fontgenerator при попытке генерировать шрифт на python 3.x выдаёт ошибки (кажется синтаксические), а на python 2.x (который рекомендован) выдаёт мол не может импортировать модуль из модулей (как-то так). В общем попробую на другом компе, или из-под linux (судя по Вашим примерам, его и используете). Ещё раз спасибо, теперь хотя-бы знаю как это будет работать. Готовые шрифты сюда приложу

lexus2k, 2020-04-04:
Unfortunately, I didn't find freetype library for Python3. As `print`, `import`, `unicode` issues could be fixed to run scripts under either Python2.7 or Python3, the problem is freetype library, which is available only for Python2.7
If you have any ideas, let me know.

Спасибо

Voha888, 2020-04-07:
Hello Aleksei. Sorry, but  I have some problems.
On python 2.7, font generation is not working for me. At 3.8 it works for eng_font.cpp, but for rus_font.cpp there are a couple of nuances

Here the font is not generated:

> > ./fontgenerator.py --ttf consola.ttf -s 8 -g 191 0 -g А 63 -f old -d -t Привет > rus_font.cpp

```
Traceback (most recent call last):
  File "fontgenerator.py", line 179, in <module>
    font.generate_fixed_old()
  File "fontgenerator.py", line 72, in generate_fixed_old
    print("// char '%s' (0x%04X/%d)" % (char, char_code, char_code))
  File "C:\Program Files\Python38\lib\encodings\cp1251.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\xbf' in position 9: character maps to <undefined>
```

But in the utf-8 variant the font is created, 

> > ./fontgenerator.py --ttf consola.ttf -s 8 -g А 63 -f old -d -t Привет > rus_font.cpp
> 

but with one problem:  **first char is 0x410**
Perhaps the font format does not allow the first character to be specified more than 0xFF (255), or am I doing something wrong?

I added an attachment with utf-8 font

[rus_font.cpp.txt](https://github.com/lexus2k/lcdgfx/files/4442165/rus_font.cpp.txt)

You asked about ideas. If there is no easy fix to the problems I have described, then I will try to apply one of two primitive options.
1. Inside the lcdgfx library, encode unicode characters to hex codes (0x00-0xFF) and then transfer to the desired font (if I succeed).
2. Make a universal font, where 255 characters contain all the necessary characters of both languages, and pass all the text through a filter that will encode the utf-cyrillic into hex-codes of the universal font. (This option should work, but takes more time)

lexus2k, 2020-04-13:
Hi,

The second example, where 0x410 is present, contains the bug, because there is no way to convert unicode char to ascii code. If you know such formula, let me know. That's why I provided example with separate character 191 before rus table:
> ./fontgenerator.py --ttf consola.ttf -s 8 -g 191 0 -g А 63 -f old -d -t Привет > rus_font.cpp

As for first issue, you described, I do not observe it on my Linux (Ubuntu 18.04). Since you say "On python 2.7", I consider that you try to run the script with python 2.7, but for some reason, in your case code goes via python 3 branch.

```.py
            if sys.version_info < (3, 0):
                print("// char '%s' (0x%04X/%d)" % (char.encode("utf-8"), char_code, char_code))
            else:
                print("// char '%s' (0x%04X/%d)" % (char, char_code, char_code)) # <<< wrong branch in your case
```
Could you please log output of sys.version_info for me, when you run Python 2.7?

VICLER, 2020-06-20:
hi @lexus2k !
I have problems with generating font using fontgenerator. I have python 2.7.18 installed and I get: ImportError: No module named freetype

GLCD Font Creator doesnt work properly for your library. It works only if the font height is <=8, but it is also mirrored to x as well to y axis.

lexus2k, 2020-06-23:
Hello, for freetype error issue you need to install library separately. That is said in fontgenerator.py script:

```.txt
# Linux:
#     Fixing pip
#     curl https://bootstrap.pypa.io/get-pip.py | python2
#     sudo pip install freetype-py
# Win32:
#     pip install freetype_py-2.0.0.post6-py2.py3-none-win32.whl
```

closed 2020-07-18 by lexus2k