lexus2k/lcdgfx #14: Cyrillic characters are not displayed correctly

Vecheslav255, 2020-04-07:
Good afternoon!
I need to display Russian text.
I am trying to use the free_calibri11x12_cyrillic font from your library kit.
The test code is very simple.

#include "lcdgfx.h"
DisplaySSD1306_128x64_I2C display(-1);
// DisplayIL9163_128x128x16_SPI display(3,{-1, 4, 5, 0,-1,-1}); // Use this line for Atmega328p
void setup()
{
    display.begin();
    display.fill(0x00);
    display.setFixedFont( free_calibri11x12_cyrillic );
    display.printFixed (11,  32,  "ААААА", STYLE_NORMAL );
}
void loop()
{
} 

But I get unreadable characters on the screen. Photo attached. I tried the same on the IL9163 display, but the result was the same.

[https://www.dropbox.com/s/lhej8to9mt0va8r/IMG_20200407_102752.jpg?dl=0](url)

I am using an Arduino UNO board with SSD1306 and IL9163 displays for the test. All demos from your library compile fine and work well

Voha888, 2020-04-07:
Try my example:

Display Initialization:
```

#include <lcdgfx.h>

//portMUX_TYPE myMutex = portMUX_INITIALIZER_UNLOCKED;

DisplayST7735_128x160x16_SPI display(DISPLAY_RST_PIN,{-1, DISPLAY_CS_PIN, DISPLAY_CLK_PIN, 0,-1,-1}); // Use this line for ESP32 (VSPI)  (gpio22=RST, gpio5=CE for VSPI, gpio21=D/C)

void display_setup_function(){

display.begin();
   display.setFreeFont( free_calibri11x12, free_calibri11x12_cyrillic );

    display.fill( 0x0000 );

    display.clear();

}
```

Image output:
```
void show(int layout_x, int layout_y, const uint8_t *size, const uint8_t *data){
    //taskENTER_CRITICAL(&myMutex);
    display.drawBitmap16(layout_x, layout_y, size[0], size[1], data);
    //taskEXIT_CRITICAL(&myMutex);
}
```

Text output:
```
void text(lcdint_t xpos, lcdint_t y, String ch, EFontStyle style){
    //taskENTER_CRITICAL(&myMutex);
    display.printFixed(xpos, y, ch.c_str(), style);
    //taskEXIT_CRITICAL(&myMutex);
}
```

Vecheslav255, 2020-04-07:
I am using Arduino IDE 1.8.12 and the lcdcfx library version 1.0.3 available for download through the arduino library manager

Vecheslav255, 2020-04-07:
Thanks, I'll try now

Voha888, 2020-04-07:
And by the way, there is a full-fledged example from the author of this library:
https://github.com/lexus2k/lcdgfx/blob/master/examples/direct_draw/draw_unicode_text/draw_unicode_text.ino

Vecheslav255, 2020-04-07:
Thanks It works

#include "lcdgfx.h"

// DisplaySSD1306_128x64_I2C display(-1);
DisplayIL9163_128x128x16_SPI display(3,{-1, 4, 5, 0,-1,-1}); // Use this line for Atmega328p

void setup()
{
   display.begin();
   display.setFreeFont( free_calibri11x12, free_calibri11x12_cyrillic );
   display.fill( 0x0000 );
   display.clear();
   display.printFixed (11,  32,  "Здравствуй мир", STYLE_NORMAL );
}

void loop()
{
}
[https://www.dropbox.com/s/r3108hbvlik49iq/IMG_20200407_114521.jpg?dl=0](url)

closed 2020-04-07 by Vecheslav255