lexus2k/lcdgfx #16: Live updating ssd1306 from camera frames

robotzero1, 2020-04-29:
Hi,

I need some pointers on how to display the frame buffer from an ESP32 camera on an OLED ssd1306 screen.

The ESP32 camera can output BMP and JPEG frames. I would prefer to use JPEG because I want to save images as well. 

Maybe I could use a jpeg decoder and something like this to reduce it down https://www.reddit.com/r/esp8266/comments/ci1zbg/best_way_to_convert_images_on_esp8266_or_esp32/evadar5/ and somehow convert it to the format used by drawBitmap1?

Alternatively, maybe I could convert this to C: https://github.com/robertgallup/python-bmp2hex/blob/master/bmp2hex.py 

Thanks.

krukhlis, 2020-04-29:
First of all you should mention ssd1306 mode -- is it i2c or SPI? Also, are you going to use 2 cores of ESP32 or single one?
I doubt ssd1306 in i2c mode can be used effectively for this task. It's refresh rate is extremally  slow, you wouldn't be able to get even 15fps from ESP32 Camera with ssd1306-controlled OLEDs( especially in single core mode, when your core will have to capture Camera frames in real time in parallel with screen refreshing of i2c ssd1306).

krukhlis, 2020-04-29:
@robotzero1 You can use this code for your inspiration( pay attention, it works with Adafruit library to draw frame): http://www.limpfish.com/esp32cam/esp32_oled.ino

robotzero1, 2020-04-29:
It's i2c. I can use two cores if necessary, Doesn't need to be fast ~5fps would be fine. The code below is working but is really slow. I don't know if it's just the Adafruit library. I'll take a look at the link above properly tomorrow. It seems similar to my code except it has some dithering code.

`#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include "esp_camera.h"

#define SCL_PIN     14
#define SDA_PIN     15
#define BACKCOLOR 0x0000 // Black
#define PIXELCOLOR 0xFFFF  // White
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
#define FRAME_WIDTH 160
#define FRAME_HEIGHT 120
uint16_t pixel_color = 0;
static esp_err_t cam_err;

// CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

// Declaration for an SSD1306 display connected to I2C (SDA, SCL pins)
#define OLED_RESET     -1 // Reset pin # (or -1 if sharing Arduino reset pin)

Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

void setup() {

  Serial.begin(115200);
  Wire.begin(14, 15);

  // SSD1306_SWITCHCAPVCC = generate display voltage from 3.3V internally
  if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) { // Address 0x3D for 128x64
    Serial.println(F("SSD1306 allocation failed"));
    for (;;); // Don't proceed, loop forever
  }

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_GRAYSCALE;
  //init with high specs to pre-allocate larger buffers
  if (psramFound()) {
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

  // camera init
  cam_err = esp_camera_init(&config);
  if (cam_err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", cam_err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  s->set_framesize(s, FRAMESIZE_QQVGA2);

}

esp_err_t camera_capture() {
  //acquire a frame
  camera_fb_t * fb = esp_camera_fb_get();
  if (!fb) {
    ESP_LOGE(TAG, "Camera Capture Failed");
    return ESP_FAIL;
  }
  int i = 0;

  for (int y = 0; y < SCREEN_HEIGHT; y++) {
    for (int x = 0; x < SCREEN_WIDTH; x++) {
      i = y * FRAME_WIDTH + x; // FRAMESIZE_QVGA // 320x240
      char ch = (char)fb->buf[i];

      if (ch > 128)
        pixel_color = WHITE;
      else
        pixel_color = BLACK;

      // Draw a single pixel in white
      display.drawPixel(x, y, pixel_color);
    }
  }
  display.display();

  //return the frame buffer back to the driver for reuse
  esp_camera_fb_return(fb);
  Serial.println("Capture frame ok.");
  return ESP_OK;
}

void loop() {
  camera_capture();
}`

lexus2k, 2020-04-30:
Hello,

Even i2c display allows refresh rate at 20fps (https://github.com/lexus2k/ssd1306/wiki/Performance-tests) Everything depends on jpeg decoding library  to receive monochrome bitmap for ssd1306 display.

robotzero1, 2020-05-06:
So I have my array[x][y] of 1s and 0s that I've decoded from the jpg, What's the best approach to get this displayed on the ssd1306?

lexus2k, 2020-05-08:
ok, so, you need to convert your buffer to the format used by ssd1306 controller (it is nicely described in datasheet).
You need to create array:
newArray[(y + 7 / 8)][x], and fill it with data from your original array:

```.cpp
array[x][y]
...
uint8_t newArray[(y + 7 / 8)][x]{};
for (int i=0; i<x; i++) {
    for (int j = 0; j<y; j++) {
        newArray[j/8][i] |= array[i][j] ? (1 << (j & 7)) : 0;
    }
}
display.drawBuffer1(0,0,x,y,&newArray[0][0]);
```

robotzero1, 2020-05-09:
I have it working! It's fast as well. I just need to work out how to show the full frame (with borders) on the screen. Also maybe change to esp_jpg_decode if I can work it out.

```
#include <TJpg_Decoder.h>
#include "lcdgfx.h"
#include "esp_camera.h"

#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

int x = 128;
int y = 64;

int grey_array [128][64];
uint8_t display_array [10][128] {};
int bitmap_row = 0;
int bitmap_column = 0;

DisplaySSD1306_128x64_I2C display(-1, { -1, 0x3C, 15, 14, 0});

static esp_err_t cam_err;

// CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

void setup() {

  Serial.begin(115200);
  display.begin();
  delay(100);

  TJpgDec.setCallback(tft_output);

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  //init with high specs to pre-allocate larger buffers
  if (psramFound()) {
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

  // camera init
  cam_err = esp_camera_init(&config);
  if (cam_err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", cam_err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  s->set_framesize(s, FRAMESIZE_QQVGA2);
}

static bool tft_output(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t* bitmap)
{
  if ( y >= 64 ) {
    return 0;
  }

  for (int p = 0; p < w * h; p++) { //pixels in current tile
    uint16_t R = (((bitmap[p]) >> 11) & 0x1F) << 3; // extract R component
    uint16_t G = (((bitmap[p]) >> 5) & 0x3F) << 2; // extract G component
    uint16_t B = ((bitmap[p]) & 0x1F) << 3; // extract B component
    uint16_t gray = (R * 30 + G * 59 + B * 11) / 100; // magic formula

    if (p % 16 == 0 && p > 0) { // 16 x 16 tile
      bitmap_row++ ;
    }

    bitmap_column = p - (bitmap_row * 16);

    grey_array[x + bitmap_column][y + bitmap_row] = gray;

    if (p == 127) {
      bitmap_row = 0;
    }
  }
  return 1;
}

esp_err_t camera_capture()
{
  camera_fb_t * fb = esp_camera_fb_get();
  if (!fb) {
    ESP_LOGE(TAG, "Camera Capture Failed");
    return ESP_FAIL;
  }

  TJpgDec.drawJpg(0, 0, (const uint8_t*)fb->buf, (uint32_t)fb->len);

  for (int i = 0; i < x; i++) {
    for (int j = 0; j < y; j++) {

      int oldpixel = grey_array[i][j];
      int newpixel  = (oldpixel > 128) ? 255 : 0;
      grey_array[i][j] = newpixel;
      int quant_error   = oldpixel - newpixel;

      grey_array[i + 1][j    ] = grey_array[i + 1][j    ] + (quant_error * 7 / 16);
      grey_array[i - 1][j + 1] = grey_array[i - 1][j + 1] + (quant_error * 3 / 16);
      grey_array[i    ][j + 1] = grey_array[i    ][j + 1] + (quant_error * 5 / 16);
      grey_array[i + 1][j + 1] = grey_array[i + 1][j + 1] + (quant_error * 1 / 16);

      int black_white = 128 < grey_array[i][j] ? 1 : 0;

      display_array[j / 8][i] |= black_white ? (1 << (j & 7)) : 0;
    }
  }

  display.drawBuffer1(0, 0, x, y, &display_array[0][0]);

  memset(grey_array, 0, sizeof(grey_array));
  memset(display_array, 0, sizeof(display_array));

  esp_camera_fb_return(fb);
  return ESP_OK;
}

void loop() {
  camera_capture();
}
```

lexus2k, 2020-05-11:
Can you describe, what it means by "with borders"? Maybe, you have pictures to attach for demonstration?

robotzero1, 2020-05-11:
At the moment I have the frame size set to QQVGA2 (128x160) basically because it makes it easy to decode the top part of the frame to give me 128x64 pixels for the screen.

What I would like to do is move to QQVGA (160 × 120) on the camera and display the whole vertical frame (120px) at 64 pixels and the horizontal however it fits (about 85px) with a black border down each side.

lexus2k, 2020-05-12:
Well, you can take 3 pixels from each 4 pixels on x axis (160 * 3 / 4 = 120), and 1 pixel from each 2 pixels on y axis (128 * 1 / 2 = 64), when you fill display array.

closed 2020-07-18 by lexus2k

robotzero1, 2020-08-10:
If anyone is interested here are the results from the different resolutions. I found out later than the minimum resolution on an ov2640 for using the full sensor is CIF so the QQVGA is using part of the sensor only as well.

[image]

Or a video is here: https://robotzero.one/wp-content/uploads/2020/08/Dithering-OLED-Demo-Optimised-1.mp4

(part of this tutorial - https://robotzero.one/esp32-cam-oled-capacitive-touch-buttons/)