lexus2k/lcdgfx #17: Getting to work with ESP IDF Master & CMake

ammaree, 2020-05-13:
Hi, first time I am trying to use your library with ESP-WROVER-KIT using ILI9341 controller and 240x320 display.

I am proficient with the ESP IDF using v4.2 (master) and reasonably so with CMake but primarily with C and less so C++. What confuses me is your various repos (canvas vs lcdgfx) and the duplication of code between them. To get the simplest possible demo of text only functionality and with the 9341 controller going, where to I start, ie:
Do I start with canvas or lcdgfx?
How do I select the ILI9341 support?
How to I select the GPIO pins for the SPI interface on the ESP-WROVER-KIT ?
Which example files shall I copy from into my GUI task to replace the setup() and loop() components?
Anything to help me get going ASAP with the basic text functionality?

All help appreciated

Andre

ammaree, 2020-05-13:
HI @lexus2k 

I think I have strung something together. As a starting point I used:
1. lcdgfx as a repo, seems more recently updated than canvas
2.  ili9341_demo.ino as starting code into my task structure (removed need for owl.cpp * owl.h)
3. created a CMakeLists.txt file as below:
```
idf_component_register(	SRCS
			./src/canvas/fonts/fonts.c
			./src/canvas/canvas.cpp
			./src/canvas/font.cpp
			./src/lcd_hal/esp/esp32_spi.cpp
			./src/lcd_hal/interface.c
			./src/v2/lcd/ili9341/lcd_ili9341.cpp
			./src/v2/lcd/lcd_common.cpp
			./src/v2//nano_engine/core.cpp
			./src/v2/ssd1306_utf8.c
		    INCLUDE_DIRS
			./src
			./src/canvas
			./src/canvas/fonts
			./src/canvas/internal
			./src/lcd_hal
			./src/lcd_hal/esp
			./src/v2/lcd
			./src/v2/lcd/ili9341
			./src/v2/lcd/base
			./src/v2/nano_engine
		)
```
The point at which the compile currently stops is as follows:
```
    ili9341_240x320_spi_init(22, 5, 21);// Use this line for ESP32 (VSPI)  (gpio22=RST, gpio5=CE for VSPI, gpio21=D/C)
C:/Dropbox/devs/ws/z-components/gui/task_gui.cpp:168:5: error: 'ili9341_240x320_spi_init' was not declared in this scope
     ili9341_240x320_spi_init(22, 5, 21); // Use this line for ESP32 (VSPI)  (gpio22=RST, gpio5=CE for VSPI, gpio21=D/C)
     ^~~~~~~~~~~~~~~~~~~~~~~
```
Any suggestions for fixing this one?

lexus2k, 2020-05-13:
Hello,

First of all you don't need other libraries except lcdgfx. So, canvas repository is just only for those, who find it useful.
Next, in CMakeLists file only single include path is required: something like this:

```.cmake
idf_component_register(
    SRCS
        ./src/canvas/fonts/fonts.c
        ./src/canvas/canvas.cpp
        ./src/canvas/font.cpp
        ./src/lcd_hal/esp/esp32_spi.cpp
        ./src/lcd_hal/esp/platform.cpp
        ./src/lcd_hal/interface.c
        ./src/v2/lcd/ili9341/lcd_ili9341.cpp
        ./src/v2/lcd/lcd_common.cpp
        ./src/v2//nano_engine/core.cpp
        ./src/v2/ssd1306_utf8.c
    INCLUDE_DIRS
        ./src
)
```

And regarding `ili9341_240x320_spi_init`, where did you get this line? From [lcdgfx/examples/demos/ili9341_demo/ili9341_demo.ino](https://github.com/lexus2k/lcdgfx/blob/master/examples/demos/ili9341_demo/ili9341_demo.ino):

```.cpp
DisplayILI9341_240x320x16_SPI display(3,{-1, 4, 5, 0,-1,-1}); // Use this line for Atmega328p
//DisplayILI9341_240x320x16_SPI display(3,{-1, -1, 4, 0, -1, -1}); // FOR ATTINY
//DisplayILI9341_240x320x16_SPI display(-1,{-1, 0, 1, 0, -1, -1); // Use this line for nano pi (RST not used, 0=CE, gpio1=D/C)
//DisplayILI9341_240x320x16_SPI display(24,{-1, 0, 23, 0,-1,-1}); // Use this line for Raspberry  (gpio24=RST, 0=CE, gpio23=D/C)
//DisplayILI9341_240x320x16_SPI display(22,{-1, 5, 21, 0,-1,-1}); // Use this line for ESP32 (VSPI)  (gpio22=RST, gpio5=CE for VSPI, gpio21=D/C)
```

ammaree, 2020-05-14:
I am lost, no idea why yours is showing different from mine. Below I have attached a screenshot of the exact same file in the repo as cloned yesterday. Anyway, I have remove the offending line(s) and the compile goes through but now I am on the next problem.
```
DisplayILI9341_240x320x16_SPI display(18, { 1, { 22 } , 21, 40000000, 19, 23 } ) ; // ESP32-WROVER-KIT VSPI  RST=18
```
The ILI9341 initialization is not working on the ESP32-WROVER-KIT. I have changed the line as follows to reflect the GPIO pins as used, according to the schematic but the display only flickers a bit during initialization then remain completely dark whilst the task goes through the different demos.
 
[image]

ammaree, 2020-05-14:
@lexus2k Just an update but I got the problem solved.
It revolves around the Backlight pin not having been configured. All I did was use the following
```
gpio_set_direction(ili9341GPIO_LIGHT, GPIO_MODE_OUTPUT) ;
gpio_set_level(ili9341GPIO_LIGHT, 0) ;
```
to configure and enable the backlight on GPIO5.
The following configuration was correct to initialize the display on the ESP32-WROVER-KIT LCD:
```
DisplayILI9341_240x320x16_SPI display(18, { 1, 22, 21, 0, 19, 23 } ) ;
```
 Maybe you can add some specific documentation for the above hardware platform.

Lastly, it appears on the surface  as is there is no support for reading the values of pixel back from the display memory. I saw in the ```SPlatformSpiConfig``` structure that no member exists for the MISO pin. Is this on purpose?

lexus2k, 2020-05-15:
Hello,

I'm glad that you have finally fixed all problems. However, regarding ILI9341 display, the problem relates more to datasheet notes, than library documentation (The library doesn't  control backlight). But your idea to add some information on wiki is good. I will do.
By the way, I removed confusing lines (with c-style API) from examples on development branch, going to merge to master soon.

lexus2k, 2020-11-05:
Cmake support is added

closed 2020-11-05 by lexus2k