lexus2k/lcdgfx #21: SH1106 not working..

VICLER, 2020-06-17:
I tried to flash an example for SH1106 and the display just stay black all the time after flashing. If I flash the same example for SSD1306 it works, but corrupted because I have SH1106. Im using Arduino Micro with Atmega32u4.

lexus2k, 2020-06-17:
Hi, thank you for reporting the issue.
I will check the problem and come back to you with comments.

Also, could you please try old library version from https://github.com/lexus2k/ssd1306 and let me know if sh1106 demo works for you there?

VICLER, 2020-06-18:
thank you for fast response!
sh1106 demo from the old library works as expected!
I would like to add some cyrillic fonts, how can I do that?

lexus2k, 2020-06-19:
You can try to generate font using `./fontgenerator.py --ttf consola.ttf -s 6 -f new -g А я` line.
Still looking into issue with displays

lexus2k, 2020-06-20:
Hello,

I found a bug in the library, and hope that my latest commit fixes it.
Could you please check lcdgfx library once again?

Thank you

VICLER, 2020-06-20:
It works now! Thank you!

closed 2020-06-25 by lexus2k