lexus2k/lcdgfx #22: SSD1306 I2C freezes

srxdevnet, 2020-06-23:
Greetings from Argentina. I´m developing on ESP8266 platform. Sometimes I´ve noticed that the display freezes and the only way to make it work again is turn off and turn on again. The freezes sometimes correspond to ESP8266 firmware flashing or ESP8266 reset.
There is a way to reset the SSD1306 using I2C bus by software?

Thanks in advance and forgive my horrible english!

lexus2k, 2020-06-23:
Hello,

1. Which IDE do you use? Esp idf?
2. Do you flash firmware via UART?
3. Do you use UART 2 USB converter? 
4. Could you please check power for your esp8266 board? I saw some freezes if power is not enough. 
5. Which speed does i2c run on? 400kHz? Can you check 100kHz mode?
6. Do you have logic analyzer to check i2c lines?
7. Do you use pullup resistors on i2c lines?

So many questions:)

srxdevnet, 2020-06-23:
Hello and thanks for your fast answer!! I´ll try to respond in the same order:

1. I´m developing with PlatformIO for Arduino on ESP8266 board ESP12e.
2. I´m flashing via UART and OTA (I´ll test if in OTA have the issue).
3. The board is a NodeMCU Amica with CP2102 converter
4. I haven´t noticed power issues at the moment.
5. The I2C speed is 400Khz, I´ll check with 100Khz.
6. No logic analizer, only oscilloscope.
7. Oh-Oh, I missed the pullup resistors!!!

Please, let me check first the pullup ressistors, and then the I2C speed.

Thank you very much for your help!

lexus2k, 2020-07-02:
Hello.
Any updates on the issue?

srxdevnet, 2020-07-02:
Hello. I apologize by my late answer.
I've added the pull up resistirá and changes yo 100khz bus speed... No changes.
I'll test updating with OTA, but it seems the issue occirs on normal operación too... Generaly after one or two days.
I'll test with another board (esp01 and esp07).

lexus2k, 2020-07-02:
If you have code example, that causes display freeze for ESP8266, it would be nice.
I have NodeMCU ESP8266 board and tested it with my ssd1306 display. It works for me.
Any additional info is welcome.

Thank you

lexus2k, 2020-11-05:
Closing the issue, since no any inputs for a long time.
If problem exists, feel free to reopen the issue.

closed 2020-11-05 by lexus2k