lexus2k/lcdgfx #23: SSD1351 Screen orientation

drewmo, 2020-06-24:
Can anyone point me in the direction of adding screen orientation support for the Waveshare SSD1351 RGB OLEDs? I need it for a project of mine.

Or if it's not possible as there is some hardware limitation that would be good to know.

I tried playing around with the library by going off the other screen orientation functions but haven't been able to make any progress.

lexus2k, 2020-06-25:
Hello,

ssd1351 has hardware options for flexible layout. It works similar way the ssd1331 works.
But I didn't add this support to lcdgfx library yet.

drewmo, 2020-06-25:
Would it be possible for me to use a similar function as used on the SSD1331? If so I'll probably look into copying it.

lexus2k, 2020-06-26:
Please check latest v1.0.0_dev branch.

lexus2k, 2020-07-02:
Hello, do you have any updates?

closed 2020-07-18 by lexus2k