lexus2k/lcdgfx #24: putPixel deletes Pixels

ACDCLabs, 2020-07-06:
the putPixel method of the NanoDisplayOps4 class does not allow to draw directly neighboring pixels. A full line appears as dotted line. 

`#include "lcdgfx.h"

DisplaySSD1327_128x128_SPI display(3, { -1, 4, 5, 0, -1, -1}); // Use this line for Atmega328p (3=RST, 4=CE, 5=D/C)

void setup() {
  display.begin();
  display.fill(0x00);

}

void loop() {

  for (int i = 0; i < 8; i++) {
    display.putPixel(64 + i, 61);
    delay(500);
  }
}`

lexus2k, 2020-07-06:
Well, actually that is not a bug.

Here you're using direct API. This API doesn't use any memory buffer, and sends all data directly to ssd1327 controller. The minimum packet size is 1 byte (8 bits). Serial interfaces: i2c, spi - do not allow to read data back from display. Each pixel for ssd1327 controller is 4 bits. So, cpu can send 2 pixels at once. Usually, the problem is solved by reading byte from memory, modifying it and saving back to memory, but this cannot be done for serial interfaces.

So, to workaround the problem, you described, you need to use either `drawLine` or canvas.

ACDCLabs, 2020-07-07:
Thanks for the answer. I will go for the Canvas. Thanks. N.

closed 2020-07-07 by ACDCLabs