lexus2k/lcdgfx #25: I'm a dummy...Need full connection help

jtannenba, 2020-08-11:
Does Vcc go to +5? where does the backlight connect.  Tried to connect to vcc with 100 ohm resistor, but screen is all white.  Should there be resistors in any of the other lines?  I'm using an Uno and using this: connect LCD to D5 (D/C), D4 (CS), D3 (RES), D11(DIN), D13(CLK)

Thank you

lexus2k, 2020-08-11:
Hi there.

Yeap, you do correct connections for TFT display. According to the link you sent, your display supports both 3.3V and 5V. So, it should be ok to supply backlight with 5V via 100 Ohm resistor.

I took my old ILI9341 display and found that it's not working too. Then I checked other libraries for ILI9341 display with my Arduino Nano, and they don't work too. Maybe, my display is broken. 
As soon as I find the solution, I will come back to you.

For now, could you please connect CS line to GND?

jtannenba, 2020-08-12:
Ok  I'll have to set it up again, I did some ??? and blew my nano.

jtannenba, 2020-08-12:
Got it to work!  Had to short J1.  See: http://www.lcdwiki.com/Run_Arduino_Demo_in_spi_model
Cute!

lexus2k, 2020-08-12:
Thank you very much for the finding. I don't have soldering iron by hand right now to check my lcd.
But I can use this TFT display with ESP32 dev board, which is supplied by 3.3V.
I need to add useful information to wiki, or to main README

Thank you

jtannenba, 2020-08-12:
A 3.3v board such as a STM32 or ESP doesn't need to be jumpered.  The 5V boards either need a level shifter in between or the jumper shorted according to lcdwiki.  Yes, adding this info to README would be a great idea.

closed 2020-08-12 by jtannenba