lexus2k/lcdgfx #26: I made a font converter that converts bdf to free lcdgfx format.

mayopan, 2020-08-28:
Hi, lexus2k!
Thank you for the nice lcd driver. I like it rather than complex and huge [Adafruit-GFX-Library](https://github.com/adafruit/Adafruit-GFX-Library).
I also like [U8G2 driver](https://github.com/olikraus/u8g2) that is a similar driver especially for monochrome displays.
It has a lot of fonts, but lcdgfx has a few.

So I made a [font converter](https://github.com/mayopan/bdf2lcdgfx) for your driver.

It converts bdf font to lcdgfx c style free (= proportional) font.
Now, I can use many bdf fonts!
Check it and if you like it, please add it in tool family for others.
Thanks.

lexus2k, 2020-08-29:
Hi there,

I'm really appreciate your effort, and of course I will add link to your converter to wiki pages.

Thank you

lexus2k, 2020-08-29:
I've updated wiki page: https://github.com/lexus2k/lcdgfx/wiki/How-to-create-new-font-for-the-library

mayopan, 2020-08-29:
Thank you for your quick response!

closed 2020-08-29 by mayopan