lexus2k/lcdgfx #27: nano-engine not working as expected

mr-who2012, 2020-09-04:
**Describe the bug**
Full-screen double-buffering with NanoEngine is not possible. The following example from the README.md-file (ssd1306/src/nano_engine/README.md) is just not working:
```

//## What if not to use draw callbacks

// If you don't want to use draw callbacks in your application, but still need a power of NanoEngine, then there is one
// way for you: to use full-screen double-buffering with NanoEngine. The example, you will find below, shows how to
// use full-screen double buffering for monochrome 128x64 ssd1306 oled display. This example can be run on Atmega328p
// and more powerful micro controllers. It clears back-buffer every time engine says to redraw the frame. But you can
// preserve previously prepared image by removing call to `engine.canvas.clear()`.

#include "ssd1306.h"
#include "nano_engine.h"

//NanoEngine<BUFFER_128x64_MONO> engine; // compile-error: BUFFER_128x64_MONO not found
NanoEngine8 engine; //Sketch compiles with this

void setup()
{
    // Init SPI 128x64 monochrome oled.
    // 3 - RESET, 4 - CS (can be omitted, oled CS must be pulled down), 5 - D/C
   //  ssd1306_128x64_spi_init(3, 4, 5); // deactivated becaus I use a different LCD
   st7735_128x160_spi_init (8, 10, 9);

    engine.begin();
    engine.setFrameRate(30);
}

void loop()
{
    if (!engine.nextFrame()) return;
    engine.canvas.clear();    // This step can be removed, if you don't want to clear buffer
    engine.canvas.drawRect(15,12,70,55);
    engine.display();
```

**Expected behavior**
A rectangle should be drawn onto the screen. Instead, there is a black box in the top-left corner.

**Please complete the following information:**
 - library version: lcdgfx by Alexey Dynda 1.0.5
 - LCD display type: st7735 with a resolution of 128x160 pixel and SPI-connection
 - Platform: Arduino Nano with an Atmega328p (old version)
 - IDE: Arduino-IDE on Ubuntu

**Additional context**
I also have problems using conditional statements in the draw-callback-function for the nano-engine. Draw-commands inside conditional statements dont get executed. 
I placed a full description of my problem on the [arduino-forums](https://forum.arduino.cc/index.php?topic=703414.msg4726777#msg4726777).

lexus2k, 2020-09-05:
ssd1306 is completely different library.
You've submitted issue to lcdgfx.

The example from here works: https://github.com/lexus2k/lcdgfx/wiki/Using-NanoEngine-for-systems-with-low-resources2

mr-who2012, 2020-09-07:
ssd1306 is the lcdgfx-library availalbe in the arduino-library-manager. Search for lcdgfx and ssd1306 shows up. All examples in the library use ssd1306.h as header file. lcdgfx.h doesn't exist.
The provided example in your post is the very same example, I mentioned before. It only differs in the usage of "lcdgfx.h" instead of ssd1306.h. lcdgfx.h doesn't exist.

lexus2k, 2020-09-07:
`ssd1306 is the lcdgfx-library availalbe in the arduino-library-manager`

Seriously?  ssd1306 and lcdgfx are 2 different libraries in Arduino library manager. ssd1306 has C-style API, while lcdgfx has C++ style API.
[image]

Please, check more carefully.

Next, you want to use full screen double buffer for 128x160 color display on Atmega328p uC. Atmega328p has 2KiB of RAM. How many bytes do you need for screen buffer??? I will answer 128x160x(16 bits / 8 bits) = 40960 bytes. And here the question: how to fit 40KiB buffer into 2KiB of RAM? Any ideas?

mr-who2012, 2020-09-08:
Why are you  typing-in "ssd1327" to get lcdgfx-library in the library-manager? Anyway, I will try the new lib that was never mentioned before, anywhere and report back.

lexus2k, 2020-09-08:
> Why are you typing-in "ssd1327" to get lcdgfx-library in the library-manager? Anyway, I will try the new lib that was never mentioned before, anywhere and report back.

It's just a keyword to search. You can use any other keyword. Do you know how search engine works?

mr-who2012, 2020-09-08:
yes, I type in a keyword and get the proper library. Here I have to type in something completely odd to get lcdgfx.

lexus2k, 2020-09-14:
Closing this issue. All discussion in https://forum.arduino.cc/index.php?topic=703414.msg4726777#msg4726777

closed 2020-09-14 by lexus2k