lexus2k/lcdgfx #28: Adding brightness support to SSDXXXX

prelard812, 2020-09-10:
I'm using a SSD1531 display and the LCDGFX with a ESP32 and the linux  simulator, with a lot of fun... Your lib is really good. :)

I can't see support to set the brightness of the display....   I would like to get it. 

I'm glad to offer my support to add it but I would like you to show me where to start, It will be easier and I will be sure that I put the function in the good Class.  

thank you.

lexus2k, 2020-09-10:
Greetings.

Unlike C-style ssd1306 library, the code for the lcdgfx displays is generated automatically. So to do that, you need to have Python installed. I made support for 2.7.X and 3.X Python, but if you have issues with any of them, let me know.
Next
1. Go to lcdgfx/tools/templates/lcd/ssd1351/functions/ and add new folder setBrightness (this will be your function)
2. Inside created folder, please create 3 files: code, doc, decl:

decl - this is declaration or function type and arguments
```.txt
void
uint8_t brightness
```
doc file should contain description
```.txt
    /**
     * @brief Sets screen brightness
     *
     * Sets screen brightness
     * @param brightness new brightness for the display
     */
```
code file should contain code for new display function
```.txt
    // To start communication with display in command mode
    this->start();
    setDataMode(0);
    // To send special command
    this->send( ??? ); // This is special command according to ssd1351 datasheet
    // To send command arguments
    setDataMode(1); // Switch to data mode to send arguments
    this->send( brightness );
    // to stop communication with display and go back to data mode
    this->stop();
```

3. Edit lcdgfx/tools/templates/lcd/ssd1351/ssd1351.json file and add `setBrightness` to `"interface_list"`, add new `setBrightness` empty section to `"functions"`.

4. Run Python script from lcdgfx/tools folder: `./lcd_code_generator.py -c all`

That's it.
Now you can call from your code
```.cpp
    display.getInterface().setBrightness(100);
```

Best regards.

prelard812, 2020-09-11:
Oh seems easy... very good integration, I will be happy to get a try.

lexus2k, 2020-09-11:
General idea, functions accessible via `display.` (`display.drawRect`) are common for all displays. 
And function accessible via `display.getInterface()` are display specific.

lexus2k, 2020-11-08:
@prelard812 How are your results? Have you succeeded in brightness control?

prelard812, 2020-11-09:
> This email originated from outside of your organization. Please do not click on links or open attachments unless you recognize the sender and know the content is safe.
> Ce courrier électronique provient de l’extérieur de l’organisation. Ne cliquez pas sur les liens ni n’ouvrez aucune pièce jointe, sauf si vous reconnaissez l’expéditeur et savez que le contenu est sécurisé.

Hi,
unfortunately I change the used screen and the LIB used… SO it will not be done. Sorry ☹

lexus2k, 2020-11-09:
Thank you.
I will leave this issue open. Maybe someday, I will implement brightness control.

simon-77, 2024-04-29:
Hey there

I currently try to use the [_EA OLEDS102-6_](https://www.displayvisions.us/fileadmin/pdf/grafik/oleds102-6.pdf) display. And as I would like to slightly adapt the implementation to better suite this display, I stumbled across this thread but have an issue following it.

First things first:
## Issue
This OLEDS102-6 display uses the SSD1306B controller - which is very similar to the already implemented SSD1306 controller.
Nevertheless, there are some issues regarding the init sequence when using the current `DisplaySSD1306_128x64_SPI` implementation:
1) the charge pump is not activated, therefore the display is dimmer than supposed to be.
2) some clock divider setup registers make the display much slower than with the example init sequence from the datasheet.

## aproach
The `lcd_ssd1306` class uses templates for different screen resolutions that also differ in the `..._initData[]` array. I therefore though about adding another template for this specifi OLED display.

I found the definition of these arrays in the JSON file:
`lcdgfx/tools/templates/lcd/ssd1306/ssd1306.json`

and therefore added to the bottom another entry called `OLEDS102_g`:
```
...
               "128x64":
                {
                    "init":
                    [
                        "#ifdef SDL_EMULATION",
                        "    SDL_LCD_SSD1306, 0x00,",
                        "    0x00, 0x00,",
                        "#endif",
                        "    0xAE, 0x00,          // display off",
                        "    0x20, 0x01, 0x00,    // Page horizontal Addressing mode",
                        "    0xC8, 0x00,          // Scan from 127 to 0 (Reverse scan)",
                        "    0x40| 0x00, 0x00,    // First line to start scanning from",
                        "    0x81, 0x01, 0x7F,    // contast value to 0x7F according to datasheet",
                        "    0xA0| 0x01, 0x00,    // Use reverse mapping. 0x00 - is normal mapping",
                        "    0xA6, 0x00,          // Normal display",
                        "    0xA8, 0x01, 63,      // Reset to default MUX. See datasheet",
                        "    0xD3, 0x01, 0x00,    // no offset",
                        "    0xD5, 0x01, 0x80,    // set to default ratio/osc frequency",
                        "    0xD9, 0x01, 0x22,    // switch precharge to 0x22 // 0xF1",
                        "    0xDA, 0x01, 0x12,    // set divide ratio",
                        "    0xDB, 0x01, 0x20,    // vcom deselect to 0x20 // 0x40",
                        "    0x8D, 0x01, 0x14,    // Enable charge pump",
                        "    0xA4, 0x00,          // Display resume",
                        "    0xAF, 0x00,          // Display on"
                    ]
                },
                "OLEDS102_6":
                {
                    "init":
                    [
                        "    0x40,                // Set Display start line ",
                        "    0xA0,                // Bottom View no Segment remap",
                        "    0xC0,                // Bottom View COM scan direction normal",
                        "    0x2E,                // StartColumnAddress",
                        "    0x8D, 0x95,          // Switch Charge Pump (9V)",
                        "    0x20, 0x02,          // Set Memory AddressMode",
                        "    0x81, 0xFF,          // Set Brightness",
                        "    0xD5, 0x40,          // Set Display Clock Divide",
                        "    0xD9, 0xF1,          // Set Precharge Periode",
                        "    0xAD, 0x30,          // Set Internal Ref",
                        "    0x21, 0x0D, 0x72,    // Set ColumnAddress",
                        "    0x22, 0x00, 0x3F,    // /Set PageAddress",
                        "    0xAF,                // Display on"
                    ]
                }
            }
        }
    }
}
```

But when runnign the python script, I do get an "list index out of range" error:
```
$ python3 lcd_code_generator.py -c all
Traceback (most recent call last):
  File "lcd_code_generator.py", line 338, in <module>
    generate_controller_data( _ctl + "/" + _ctl + ".json", _ctl)
  File "lcd_code_generator.py", line 314, in generate_controller_data
    g_voc["height"] = res.split('x')[1]
IndexError: list index out of range
```

Is there another place where I have to "add" this new configuration option "OLEDS102_6"?
Do you know what else might cause this issue?

Best regards,
Simon

PS.
Another thing I have noticed about this particular display: it has a pixel size of "only" 102x64 out of the 128x64.
To put Text on the left side of the Display frame I have to provide an offset of 12 pixels, e.g. by: `.printFixed(12.0. "My Text")`.
It would be great to also adapt this offset for the implementation of this display.

simon-77, 2024-05-12:
Any chance to get some hints? @lexus2k 

Or would it be better to open a separate issue?