lexus2k/lcdgfx #29: Missing documentation

myval, 2020-10-29:
Not bug as such in the library but the link to the documentation is not working.

lexus2k, 2020-10-29:
Could you please describe, which link doesn't work. 
https://codedocs.xyz/lexus2k/lcdgfx

myval, 2020-10-29:
yeah sorry should have specify better. The link in the documentation section here:

Documentation generated by doxygen tool can be found at lcdgfx github.io site. 
this link:
`http://lexus2k.github.io/lcdgfx`

myval, 2020-10-29:
Yeah the one you sent is fine, thanks. The link in document section might need to be updated.

lexus2k, 2020-10-29:
Thank you, now fixed.
Could you check please ?

myval, 2020-10-29:
Works first class now, thanks.
while I got you on line can you provide an example for printFixedPgm()? I have been struggling with it a bit.
And second question what is the function to set cursor for lets say print()?

thanks

lexus2k, 2020-10-30:
> while I got you on line can you provide an example for printFixedPgm()?
Which platform? AVR?

```.cpp
const char * PROGMEM text = "Text from flash memory"
...
display.printFixedPgm(0,0, text);
```

> And second question what is the function to set cursor for lets say print()?

`display.setCursor()`. Does it work for you?

closed 2020-11-02 by lexus2k