lexus2k/lcdgfx #30: Example lode_runner_ili9341.ini don't compile, missing method setRotation

lcgamboa, 2020-11-07:
Using Arduino 1.8.13 IDE the example lode_runner_ili9341.ini don't compile when line 286 is uncommented to enable display rotation. https://github.com/lexus2k/lcdgfx/blob/ef1a62fd4c86a7f5421c37065b6528634573f677/examples/games/lode_runner_ili9341/lode_runner_ili9341.ino#L286

Error:
```
/tmp/arduino_modified_sketch_394812/lode_runner_ili9341.ino: In function 'void setup()':
lode_runner_ili9341:286:13: error: 'GraphicsDisplay {aka class DisplayILI9341_240x320x16_SPI}' has no member named 'setRotation'
     display.setRotation(1);
             ^~~~~~~~~~~
exit status 1
'GraphicsDisplay {aka class DisplayILI9341_240x320x16_SPI}' has no member named 'setRotation'
```
Using:
 - library version : 1.0.5
 - LCD display type: ili9341
 - OS: linux
 - Platform: Atmega328p
 - IDE: Arduino  1.8.13

lexus2k, 2020-11-08:
Thank you for reporting the issue. 
Try this line instead:

```.cpp
    display.getInterface().setRotation(1);
```

lcgamboa, 2020-11-08:
Thank you, problem solved.

closed 2020-11-08 by lcgamboa