lexus2k/lcdgfx #31: Can't get it to compile - can you help?

sixtyfive, 2020-11-10:
Tried both from the Library Manager and cloned from GitHub; with the latter it looks like this:

```
/usr/share/avr/bin/avr-ld: /tmp/cct12l2x.ltrans0.ltrans.o: in function `NanoEngineInputs::ky40Buttons()':
/home/jrs/Arduino/libraries/lcdgfx/src/v2/nano_engine/core.cpp:141: undefined reference to `NanoEngineInputs::s_ky40_clk'
/usr/share/avr/bin/avr-ld: /home/jrs/Arduino/libraries/lcdgfx/src/v2/nano_engine/core.cpp:141: undefined reference to `NanoEngineInputs::s_ky40_clk'
/usr/share/avr/bin/avr-ld: /home/jrs/Arduino/libraries/lcdgfx/src/v2/nano_engine/core.cpp:141: undefined reference to `NanoEngineInputs::s_ky40_dt'
/usr/share/avr/bin/avr-ld: /home/jrs/Arduino/libraries/lcdgfx/src/v2/nano_engine/core.cpp:141: undefined reference to `NanoEngineInputs::s_ky40_sw'
/usr/share/avr/bin/avr-ld: /tmp/cct12l2x.ltrans0.ltrans.o: in function `NanoEngineInputs::connectKY40encoder(unsigned char, unsigned char, signed char)':
/home/jrs/Arduino/libraries/lcdgfx/src/v2/nano_engine/core.cpp:133: undefined reference to `NanoEngineInputs::s_ky40_clk'
/usr/share/avr/bin/avr-ld: /home/jrs/Arduino/libraries/lcdgfx/src/v2/nano_engine/core.cpp:133: undefined reference to `NanoEngineInputs::s_ky40_dt'
/usr/share/avr/bin/avr-ld: /home/jrs/Arduino/libraries/lcdgfx/src/v2/nano_engine/core.cpp:133: undefined reference to `NanoEngineInputs::s_ky40_sw'
collect2: fatal error: ld returned 1 exit status
compilation terminated.
exit status 1
Error compiling for board Arduino Nano.
```

My only code so far was:

```cpp
#include "lcdgfx.h"

void setup(){}
void loop(){}
```

Using `<lcdgfx.h>` instead of the quotation marks also doesn't help :-(

lexus2k, 2020-11-10:
Hello,

If you use Arduino IDE, then it doesn't need any Makefile. So, there are many open questions to you.

The example, you've pointed above, is compiled successfully for Arduino Nano.
```.cpp
#include "lcdgfx.h"

void setup(){}
void loop(){}
```

1. What is hardware you try to compile for?
2. Do you use Arduino IDE or some different IDE?
3. What is version of IDE?

When you submit new issue, there is detailed description of all fields, needed to be filled.

lexus2k, 2020-11-11:
Anyway, it would be nice if you enable extended logs in Arduino IDE.

sixtyfive, 2020-11-11:
Hello @lexus2k,

>     1. What is hardware you try to compile for? **Arduino Nano**.

correct, Arduino Nano.

>     2. Do you use Arduino IDE or some different IDE?

Arduino IDE.

>     3. What is version of IDE?

1.8.13.

>     4. Do you have ssd1306 library installed?

I had, but I completely removed everything else having to do with OLEDs. The only such library is now `lcdgfx`.

> Anyway, it would be nice if you enable extended logs in Arduino IDE.

Of course, gladly! The log is here: https://gist.github.com/sixtyfive/d3ecebd7132607412ec767e6bd2e367c

lexus2k, 2020-11-11:
It looks like you use new version of avr-gcc more strict compiler. So, the fix is on it's way.

lexus2k, 2020-11-11:
I made the fix e62eb15. Could you please get the latest version from v1.0.0_dev branch and verify it?

sixtyfive, 2020-11-13:
That was quick, wow! I tried it again and it compiles and links fine now. Unfortunately it doesn't fit onto the board:

```cpp
#include <lcdgfx.h>

void setup(){}
void loop(){}
```

```
Linking everything together...
/usr/share/avr/bin/avr-gcc -Wall -Wextra -Os -g -flto -Wl,--gc-sections -mmcu=atmega328p -o /tmp/arduino_build_728714/CapacityTester.ino.elf /tmp/arduino_build_728714/sketch/CapacityTester.ino.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/canvas/canvas.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/canvas/font.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/canvas/fonts/fonts.c.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/interface.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/arduino/arduino_spi.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/arduino/arduino_wire.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/arduino/platform.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/avr/i2c_embedded.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/avr/i2c_twi.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/avr/platform.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/avr/spi_avr.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/avr/spi_usi.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/esp/esp32_i2c.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/esp/esp32_spi.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/esp/esp8266_i2c.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/esp/esp8266_spi.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/esp/platform.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/linux/linux_i2c.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/linux/linux_spi.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/linux/platform.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/linux/sdl_i2c.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/linux/sdl_spi.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/mingw/platform.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/lcd_hal/stm32/platform.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/utf8.c.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/gui/button.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/gui/menu.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/gui/yesno.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/lcd_common.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/il9163/lcd_il9163.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/ili9341/lcd_ili9341.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/lcdany/lcd_any.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/pcd8544/lcd_pcd8544.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/sh1106/lcd_sh1106.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/sh1107/lcd_sh1107.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/ssd1306/lcd_ssd1306.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/ssd1325/lcd_ssd1325.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/ssd1327/lcd_ssd1327.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/ssd1331/lcd_ssd1331.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/ssd1351/lcd_ssd1351.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/lcd/st7735/lcd_st7735.cpp.o /tmp/arduino_build_728714/libraries/lcdgfx/v2/nano_engine/core.cpp.o /tmp/arduino_build_728714/libraries/SPI/SPI.cpp.o /tmp/arduino_build_728714/libraries/Wire/Wire.cpp.o /tmp/arduino_build_728714/libraries/Wire/utility/twi.c.o /tmp/arduino_build_728714/../arduino_cache_298811/core/core_arduino_avr_nano_cpu_atmega328_1621df717313d057c92202babd71649a.a -L/tmp/arduino_build_728714 -lm
/usr/share/avr/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_728714/CapacityTester.ino.elf /tmp/arduino_build_728714/CapacityTester.ino.eep
/usr/share/avr/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_728714/CapacityTester.ino.elf /tmp/arduino_build_728714/CapacityTester.ino.hex
Using library lcdgfx at version 1.0.5 in folder: /home/jrs/.arduino15/sketches-and-libraries/libraries/lcdgfx 
Using library SPI at version 1.0 in folder: /usr/share/arduino/hardware/arduino/avr/libraries/SPI 
Using library Wire at version 1.0 in folder: /usr/share/arduino/hardware/arduino/avr/libraries/Wire 
/usr/share/avr/bin/avr-size -A /tmp/arduino_build_728714/CapacityTester.ino.elf
Sketch uses 32456 bytes (105%) of program storage space. Maximum is 30720 bytes.text section exceeds available space in board

Global variables use 368 bytes (17%) of dynamic memory, leaving 1680 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Arduino Nano.
```

sixtyfive, 2020-11-13:
Out of a hunch I now tried the Adafruit library, and that one exhibits the same size problem. But I can remember having used it without trouble a few years back. Another thing to do with the changes in compiler version, perhaps?

lexus2k, 2020-11-13:
Adafruit library needs much more memory since it is implemented by classes with virtual methods. All virtual methods are usually included in binary release.
Can you share the sketch, which doesn't fit (or write similar)?
Also, please, note that lcdgfx has UserSettings.h header, which allows to disable not required functionality.
And one more thing, useful for debugging: you can run `/usr/share/avr/bin/avr-nm --size-sort -C -r /tmp/arduino_build_728714/CapacityTester.ino.elf` to get listing of all functions and their size in flash memory

sixtyfive, 2020-11-14:
>  Can you share the sketch, which doesn't fit (or write similar)?

It's the same I already shared; difficult to make it any smaller:

```cpp
#include <lcdgfx.h>

void setup(){}
void loop(){}
```

>  Also, please, note that lcdgfx has UserSettings.h header, which allows to disable not required functionality.

Ah, I'd been looking for something like this, but in the project root, not in the deeper folders.  With the following configuration:

```cpp
//#define CONFIG_SOFTWARE_I2C_ENABLE
#define CONFIG_TWI_I2C_ENABLE
#define CONFIG_AVR_SPI_ENABLE
//#define CONFIG_USI_SPI_ENABLE
//#define CONFIG_AVR_UART_ENABLE
//#define CONFIG_VGA_ENABLE
//#define CONFIG_ADAFRUIT_GFX_ENABLE
#define CONFIG_PLATFORM_I2C_ENABLE
//#define CONFIG_PLATFORM_SPI_ENABLE
//#define CONFIG_LINUX_I2C_ENABLE
//#define CONFIG_LINUX_SPI_ENABLE
#define CONFIG_ARDUINO_I2C_ENABLE
//#define CONFIG_ARDUINO_SPI_ENABLE
//#define CONFIG_ESP32_I2C_ENABLE
//#define CONFIG_ESP32_SPI_ENABLE
//#define CONFIG_ESP8266_I2C_ENABLE
//#define CONFIG_ESP8266_SPI_ENABLE
//#define CONFIG_SSD1306_UNICODE_ENABLE
```

it now gives this output:

```
/usr/share/avr/bin/avr-size -A /tmp/arduino_build_673379/sketch_nov14c.ino.elf
Sketch uses 30624 bytes (99%) of program storage space. Maximum is 30720 bytes.
Global variables use 355 bytes (17%) of dynamic memory, leaving 1693 bytes for local variables. Maximum is 2048 bytes.
```

Okay, better than before! What's taking up so much space?

```bash
IFS=$'\n'; for line in $(/usr/share/avr/bin/avr-nm --size-sort -C -r /tmp/arduino_build_673379/sketch_nov14c.ino.elf); do IFS=' ' read hex m file <<<"$line"; echo -e "$((16#$hex))\t$file"; done | head -n 20
3076	comic_sans_font24x32_123
1524	ssd1306xled_font8x16
1219	free_calibri11x12_latin
944	free_calibri11x12_cyrillic
859	free_calibri11x12
738	NanoCanvasOps<(unsigned char)1>::drawBitmap1(int, int, unsigned int, unsigned int, unsigned char const*)
731	courier_new_font11x16_digits
646	NanoCanvasOps<(unsigned char)4>::drawBitmap1(int, int, unsigned int, unsigned int, unsigned char const*)
581	ssd1306xled_font6x8
576	__vector_24
554	NanoCanvasOps<(unsigned char)16>::drawBitmap1(int, int, unsigned int, unsigned int, unsigned char const*)
546	NanoCanvasOps<(unsigned char)4>::drawBitmap8(int, int, unsigned int, unsigned int, unsigned char const*)
490	NanoCanvasOps<(unsigned char)1>::drawVLine(int, int, int)
485	ssd1306xled_font5x7
484	digital_font5x7
482	NanoCanvasOps<(unsigned char)8>::drawBitmap1(int, int, unsigned int, unsigned int, unsigned char const*)
480	twi_writeTo
452	twi_readFrom
395	ssd1306xled_font6x8_AB
394	NanoCanvasOps<(unsigned char)1>::fillRect(int, int, int, int)
```

Fonts, that's expected, but so far couldn't find a wat to disable them. What else, perhaps SPI (I don't need it):

```bash
$ IFS=$'\n'; for line in $(/usr/share/avr/bin/avr-nm --size-sort -C -r /tmp/arduino_build_673379/sketch_nov14c.ino.elf); do IFS=' ' read hex m file <<<"$line"; echo -e "$((16#$hex)) $file"; done | grep -i spi | cut -d ' ' -f 1 | awk '{s+=$1} END {print s}' -
4414
```

Wow, quite a chunk! Unfortunately, it's not possible to disable SPI. With:

```cpp
//#define CONFIG_AVR_SPI_ENABLE
```

it doesn't compile at all:

```bash
[...]
$ /usr/share/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/eightanaloginputs -I/home/jrs/.arduino15/sketches-and-libraries/libraries/lcdgfx/src -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/src /home/jrs/.arduino15/sketches-and-libraries/libraries/lcdgfx/src/v2/utf8.c -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Error while detecting libraries included by /home/jrs/.arduino15/sketches-and-libraries/libraries/lcdgfx/src/v2/utf8.c
Using cached library dependencies for file: /usr/share/arduino/hardware/arduino/avr/libraries/Wire/src/Wire.cpp
Using cached library dependencies for file: /usr/share/arduino/hardware/arduino/avr/libraries/Wire/src/utility/twi.c
Generating function prototypes...
/usr/share/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/eightanaloginputs -I/home/jrs/.arduino15/sketches-and-libraries/libraries/lcdgfx/src -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/src /tmp/arduino_build_673379/sketch/sketch_nov14c.ino.cpp -o /tmp/arduino_build_673379/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
In file included from /home/jrs/.arduino15/sketches-and-libraries/libraries/lcdgfx/src/nano_gfx_types.h:31,
                 from /home/jrs/.arduino15/sketches-and-libraries/libraries/lcdgfx/src/lcdgfx.h:30,
                 from /home/jrs/.arduino15/sketches-and-libraries/archive/sketch_nov14c/sketch_nov14c.ino:1:
/home/jrs/.arduino15/sketches-and-libraries/libraries/lcdgfx/src/lcd_hal/io.h:581:2: error: #error "Platform not supported"
 #error "Platform not supported"
  ^~~~~
```

Sorry to keep bothering you with this, I just really want to play with lcdgfx :-(

lexus2k, 2020-11-15:
Hi, adding such options is not the solution for the problem, since API and variables do not lead to flash memory consumption by theirselves, if they are not used. Some exceptions from this rule due to C++ nature may relate to Wire and SPI libraries since they use static objects and C++ compiler doesn't know if it should initialize SPI/Wire object or not regardless on using of specific API in your sketch.
But fonts definitions, which are not used by your sketch, definitely cannot not lead to flash memory consumption. Something else is wrong with IDE in your case. 

Empty sketch with including lcdgfx, needs only 1690 bytes of flash in my case on Windows (Arduino IDE 1.8.13):
```.txt
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\Alex\\AppData\\Local\\Temp\\arduino_build_541449/sketch_nov11a.ino.elf"
Sketch uses 1690 bytes (5%) of program storage space. Maximum is 30720 bytes.
Global variables use 136 bytes (6%) of dynamic memory, leaving 1912 bytes for local variables. Maximum is 2048 bytes.
```

And 1254 bytes in Linux (Arduino IDE 1.8.10)
```.txt
/home/user/arduino-1.8.10/hardware/tools/avr/bin/avr-size -A /tmp/arduino_build_951502/sketch_nov15a.ino.elf
Sketch uses 1254 bytes (4%) of program storage space. Maximum is 30720 bytes.
Global variables use 130 bytes (6%) of dynamic memory, leaving 1918 bytes for local variables. Maximum is 2048 bytes.
```

Another moment, Usually Arduino IDE uses avr-gcc compiler from it's package, but not the one installed in the system. In my case the path to cross-tools is `/home/user/arduino-1.8.10/hardware/tools/avr/bin/avr-gcc`, and in your case is `/usr/share/avr/bin/avr-gcc`. That's talking about different install process of Arduino IDE.

Also you mentioned above that there are issue with Adafruit library in your setup.

Next, I compared gcc options, and they are different. At least my version uses additional option `-fuse-linker-plugin` (Both Windows and Linux) for linking the sketch.

Could you please check your `hardware\arduino\avr\platform.txt` platform file, used for the board?
And you can always compile any sketch from lcdgfx examples for your nano using command line script:

```.txt
cd tools
./build_and_run.sh -p avr demos/ssd1306_demo
```
This sketch needs only 10KiB.

sixtyfive, 2020-11-21:
Hi Aleksei, sooo, after another half day spent on this, I'm giving up. It has turned out that the culprit is indeed somewhere in the way my distro (Solus) is providing the Arduino IDE. I've switched to PlatformIO for the project and everything is as it's supposed to be without much hassle. Thanks to your help, there is now [a working upgrade](https://github.com/sixtyfive/arduino-battery-capacity-tester-3.1) to the already very nice Arduino Battery Capacity Tester. Looking forward to a bigger project using lcdgfx in the future very much! Cheers, Jonathan

closed 2020-11-21 by sixtyfive

lexus2k, 2020-11-23:
Hello,

Thank you for your response. It's sad that I couldn't help you with Solus and Arduino IDE. But I'm glad that PlatformIO works for you. Thank you for using lcdgfx, and bringing more open source projects

Best regards