lexus2k/lcdgfx #36: Updating "Arduino SAMD Boards" package to 1.8.10 has broken compilation

stealth-ultd, 2020-12-18:
**Describe the bug**

The update of the "Arduino SAMD Boards" package to 1.8.10 (Arduino IDE) seems to have broken compilation for the lcdgfx library ("fatal error / compilation terminated" – see below). Downgrading to package 1.8.9 removes the compiler error. Upgrading to 1.8.10 re-instates the error.

**To Reproduce**
Steps to reproduce the behavior:
1. Update to "Arduino SAMD Boards" 1.8.10 in Arduino IDE
2. Click 'Compile' in IDE
3. Compiler terminates:
"…/Arduino/libraries/lcdgfx/src/lcd_hal/arduino/io.h:42:14: fatal error fatal error: avr/pgmspace.h: No such file or directory
     #include <avr/pgmspace.h>
              ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino MKRZERO."

**Expected behavior**
Expected: no compiler error (as confirmed with SAMD Board package 1.8.9)

**Screenshots**
n/a

**Please complete the following information:**
 - library version: 1.0.6 (latest, Github)
 - LCD display type: ssd1306 128x64
 - OS: MacOS 11.1
 - Platform: MKR Zero
 - Arduino IDE 1.8.14 Hourly Build 2020/12/01

**Additional context**
n/a

lexus2k, 2020-12-19:
They changes some things in their release, hope that the fix above will help you.
Also, pull requests are very welcome to development branch: [v1.0.0_dev](https://github.com/lexus2k/lcdgfx/tree/v1.0.0_dev)

stealth-ultd, 2020-12-19:
Thanks for the quick action, it compiles like a charm, all fixed!

closed 2021-01-07 by lexus2k