lexus2k/lcdgfx #37: positiveMode/ negativeMode and innitialization fails

myval, 2021-01-04:
Hi,
I am guessing it is a bug. I recompiled old code of mine which used to work fine now I am getiitng this:
'class DisplaySSD1306_128x64_I2C' has no member named 'negativeMode'

Have I missed any changes or has it been dorpped in the latest update?

Second issue not sure if again related to lcdgfx update when I turn it on first time it does not initialize display correctly I end up with randome pixles on the screen. When I turn power off and on it initializes correctly.

I am using the same hardware I used while back and the same arduino sketch but compiling in now gives me these two problems.

Thanks
Josef

lexus2k, 2021-01-04:
@myval  Thank you for reporting.

> I am guessing it is a bug. I recompiled old code of mine which used to work fine now I am getiitng this: 'class DisplaySSD1306_128x64_I2C' has no member named 'negativeMode'

Use `invertColors()` method now - it will switch between 2 modes , or if you want to control by yourself use `setBackground()` and `getColor()` methods.

> Second issue not sure if again related to lcdgfx update when I turn it on first time it does not initialize display correctly I end up with randome pixles on the screen. When I turn power off and on it initializes correctly.

I didn't change anything. More details please

myval, 2021-01-05:
Thanks, I will try that hopefully soon and report back. 

I will try to gather mor details about that ini issue but it might take few days, just got another project in the way to finish.

myval, 2021-01-27:
I am going to get to it hopefully end of this week. I got heavily sidetracked on another project.

myval, 2021-02-08:
Finally got bit of a time to get back to this project and I have found out that the problem is with the Oleds actually. I am using SPI oleds but I reconfigure them to I2C and that is where the problem lies I just discovered since they need (I am guessing little bit of a delay on start up on the RES line. The I2C Oleds use RC to on the RES line but the SPI have nothing like that  in place. I have used some online info on how to convert them but I see it is wrong and I have to add RC on the RES line to get it to work.
I will do bit more investigation tomorrow but I am guessing it is the issue.

lexus2k, 2021-02-16:
Hi @myval ,

The most displays with SPI I saw, require using of RES line. 
Have you succeeded in modifying LCD displays? 

Best regards

myval, 2021-02-16:
Yes I got it sorted I had to add an RC on the RES line to have bit of a delay on the line here are two pics of how I got it done. I scraped little area on the pcb and soldered resistor to res line and capacitor between res line and ground and it works like a charm now.

[image]

[image]

lexus2k, 2021-02-20:
Thank you for the pictures. They will be very helpful for many developers.

Feel free to reopen the issue, if the problem still persists.

closed 2021-02-20 by lexus2k