lexus2k/lcdgfx #38: Which python version is compatible for fontgenerator?

Danixu, 2021-01-17:
Hello,

I'm trying to convert an Arial font just for testing, because I need some icons that are not in included fonts like for example ►. In the first attemp I've received an error saying that a library is missing, but after installing that library I get this error:

```
Traceback (most recent call last):
  File "C:\Users\danix\Desktop\Fuentes\fontgenerator.py", line 187, in <module>
    font.generate_new_format()
  File "C:\Users\danix\Desktop\Fuentes\modules\fontgenerator.py", line 117, in generate_new_format
    print("// char '%s' (0x%04X/%d)" % (char, ord(char), ord(char)))
  File "C:\Users\danix\AppData\Local\Programs\Python\Python39\lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\x80' in position 9: character maps to <undefined>
```

I'm using python 3.9.1 on Windows, so that is why I want to ask the compatible versions, because I've only seen 2,7 and 3, but not if works in the latest.

Best regards, and thanks for this library. I was searching for a lightweight library for my project because the Adafruit library is very big.

Danixu, 2021-01-17:
Testing I've found that a letter on that font seems to be incompatible. Removing that letter (80), it seems to works fine. Also is not compatible with python3 just because a funcition used that is deprecated. The function is unichr that is replaced by chr.

Best regards.

lexus2k, 2021-01-18:
Hi @Danixu,

Hmmm,... I have python 3.8.5 installed and do not observe any issues. Maybe the problem relates to operating system, but not the Python version

lexus2k, 2021-01-18:
Just installed Python 3.9.1 for Windows 10, and there are now any issues with the command, pointed above:
`python fontgenerator.py --ttf arial.ttf -s 8 -fw -g 10 255 -f new -d`

Danixu, 2021-01-18:
Hello @lexus2k, thanks for your response.

Is strange... now that command is working fine on my PC. It just fails when I try to redirect the stdout to a file, but I've modified the tool to save directly to a file and is working when I set the encoding to "UTF-8". The problem comes when I open that file on an editor, because all symbols are wrong. I've tried the function sys.stdout.buffer.write and same behaviour, the pipe output is fixed and now I can send the stdout data to a file, but the symbols are not shown in code editors.

I'll do more tests to see, because that files looks fine on the console and only seems to be wrong on editors.

Best regards

lexus2k, 2021-01-18:
@Danixu  Yes, some symbols are non-printable. As a solution, I could limit printing to only ASCII chars (0-127) and uni-codes only (more than 255).

Danixu, 2021-01-18:
would be nice to remove some non-printable characters, because some of them can break the output file. For example char 13 is a line break, char 32 is just an space that doesn't create any entry, just `// char ' ' (0x0020/32)`, and between 128 and 159 are not printable even on windows terminal.

I've modified the code to make it work even on Python 3.9.1 and Windows, and also I've added an option to save the output directly to a file (but maybe is not required anymore now that stdout to file works fine). I've done some tests on my Python 3.9.1 installed on Windows, and python 2.7/3.7 installed on my Linux Subsystem (Debian on Windows with WSL).

Best regards.

lexus2k, 2021-01-28:
Hi

I added function to filter output characters. If there are still issues with generating font tables, feel free to reopen the issue.

closed 2021-01-28 by lexus2k