lexus2k/lcdgfx #40: SSD1327 – implement BOLD and SIZE_2X?

stealth-ultd, 2021-01-22:
I have been using display.printFixedN(... FONT_SIZE_2X) and display.printFixed(..., STYLE_BOLD) with the SSD1306. However, with the larger SSD1327 this seems not to work. In short: It would be helpful when using using DisplaySSD1327_128x128_I2C display(-1) constructor such functionality can be used.

As a compromise, I now use ssd1306xled_font8x16 in combination with ssd1306xled_font6x8 on the SSD1327, but the larger 8x16 font looks quite different in style, and cannot be made bold. A pity as this larger display would benefit from these features and its quite elegant implementation.

Now I am not sure if there is an easy fix, or not at all – but I thought to bring it up in any case!

lexus2k, 2021-01-24:
Hi, @stealth-ultd 

Yeah, it is possible to implement such function (to display 2x font size). 
As for bold implementation, can you uncomment line in `lcdgfx/src/v2/lcd/base/ssd1306_4bit.inl` anc check?

```.cpp
template <class I>
void NanoDisplayOps4<I>::printFixed(lcdint_t xpos, lcdint_t y, const char *ch, EFontStyle style)
{
    // TODO: fontstyle not supported
    // m_fontStyle = style;
    this->m_cursorX = xpos;
    this->m_cursorY = y;
    while (*ch)
    {
        this->write(*ch);
        ch++;
    }
}
```

Maybe bold implementation will work somehow (I'm not sure)

stealth-ultd, 2021-01-24:
Hi Aleksei,

Thanks for getting back on this. I tried the bold implementation, but am not sure if I did exactly what you expected:

uncommented:
// m_fontStyle = style;
and replaced it with:
this->m_fontStyle = style;

It compiles OK, but does not make the font bold, and makes it a bit unreadable.

lexus2k, 2021-01-25:
@stealth-ultd  Yeah, I see the issue.

This feature cannot be implemented fast for ssd1327 display. That problem is caused by the structure of ssd1327 GDRAM. Each byte represents 2 pixels: (x,y) and (x+1,y), - horizontal row. While for ssd1306 display each byte represents: (x,y), ... ,(x,y+7) - vertical row. lcdgfx library makes bold fonts by putting char image twice with (1,0) shift. That works for ssd1306, but does not work for ssd1327.
Shifting char image by (0,1) doesn't give any positive result - the font doesn't look like bold.
So, the feature has no quick solution.

stealth-ultd, 2021-01-25:
This indeed seems a bit more complicated. Thanks for looking into this!