lexus2k/lcdgfx #41: fontgenerator.py fails with exception

julian-weinert, 2021-01-30:
I'm trying to convert a font using the included fontgenerator tool. I tried it on my Mac as well on a Debian 10 installation, using python 2.7 and also python 3.

Every time I try to convert a font the following error occurs 

```
python2.7 fontgenerator.py --ttf NiceFont-Medium.ttf -s 12 -d -f new > ~/Desktop/nice_font-12.h
Password:
Traceback (most recent call last):
  File "fontgenerator.py", line 151, in <module>
    source = ttfsource.TTFSource(fname, fsize)
  File "Arduino/libraries/lcdgfx/tools/modules/ttfsource.py", line 45, in __init__
    self.face = freetype.Face( filename )
  File "/Library/Python/2.7/site-packages/freetype/__init__.py", line 991, in __init__
    raise FT_Exception(error)
freetype.ft_errors.FT_Exception: FT_Exception:  (cannot open resource)
```

I first thought it might be a permission issue, but I tried it with sudo as well. Then I assumed it was one of the funny Apple things and switched to my Debian installation which failed in exactly the same manner.

lexus2k, 2021-01-31:
Hi @julian-weinert 

It looks like the problem with Python freetype library. I would advise you to rename font to the name without dash. If the problem still persists, submit the issue to freetype library. The following code fails to pass:

```.py
    import freetype
    font = freetype.Face( "NiceFont-Medium.ttf" )
```

Another possible root cause is the path to the font file. Please, check that the path is correct.

Best regards.

closed 2021-02-10 by lexus2k

julian-weinert, 2021-02-11:
Sorry for getting back to you so late. Yes, I agree it must be an issue with the freetype library.
It seems that its name matching is broken.

When I specifying the full path as you suggested it works. Thanks for the help