lexus2k/lcdgfx #42: displaying floating point numbers

leobg, 2021-02-02:
Ask any question, you have, regarding the library
Appology for the newbie question, I am struggling to find way to display floating point number with lcdfgx, for example on serial you could do Serial.println(1.23456, 2) to get "1.23" displayed. I couldn't find anything in the demos I could use either. Any help will be greatly appreciated!

leobg, 2021-02-02:
I should probably add I am trying to use this library with a DisplaySSD1306_128x64_I2C

krukhlis, 2021-02-02:
> I should probably add I am trying to use this library with a DisplaySSD1306_128x64_I2C

It doesn't metter which library you gonna use. Approach is the same -- first you convert using regular functions your numbers to String( here you specify formatting like rounding, number of digits after . , etc). And then you print this string using your graphics library of choice.

leobg, 2021-02-02:
> It doesn't metter which library you gonna use. Approach is the same -- first you convert using regular functions your numbers to String( here you specify formatting like rounding, number of digits [...]

Thanks @krukhlis, that's what I just actually end up doing.

closed 2021-02-02 by leobg