lexus2k/lcdgfx #43: Rotate 180 degrees a SSD1306 oled on RPI

jancotipsarevic, 2021-02-06:
Hello, I am starting using this amazing library, I am not an expert in cpp but a need to use it.
This simple example works as expected but the board I am using has the oled physically rotated 180 degrees, so I need to rotate it but I dont know how to do it. I believe it is a matter of cpp knowledge. I need to overcome this issue to follow with my tests. I'd really appreciate your help.

Thank you very much.

int main(int argc, char **argv)
{
DisplaySSD1306_128x64_I2C display((int8_t) 5, { -1, 0x3D, -1, -1, 0 });
display.begin();
display.clear();
display.drawLine(10,10,50,10);
display.setFixedFont(ssd1306xled_font6x8);
display.printFixed (0, 8, "Line 1. Normal text", STYLE_NORMAL);
cout << "Finished!, press any key to continue" << endl;
cin.get();
return EXIT_SUCCESS;
}

jancotipsarevic, 2021-02-06:
I've managed to solved it by changing the init funtion and compile a new lib.
This function is located at /lcdgfx/src/v2/lcd/ssd1306/lcd_ssd1306.inl.

`/*//ORIGINAL FUNCTION
static const PROGMEM uint8_t s_SSD1306_lcd128x64_initData[] =
{
#ifdef SDL_EMULATION
    SDL_LCD_SSD1306, 0x00,
    0x00, 0x00,
#endif
    0xAE, 0x00,          // display off
    0x20, 0x01, 0x00,    // Page horizontal Addressing mode
    0xC8, 0x00,          // Scan from 127 to 0 (Reverse scan)
    0x40| 0x00, 0x00,    // First line to start scanning from
    0x81, 0x01, 0x7F,    // contast value to 0x7F according to datasheet
    0xA0| 0x01, 0x00,    // Use reverse mapping. 0x00 - is normal mapping
    0xA6, 0x00,          // Normal display
    0xA8, 0x01, 63,      // Reset to default MUX. See datasheet
    0xD3, 0x01, 0x00,    // no offset
    0xD5, 0x01, 0x80,    // set to default ratio/osc frequency
    0xD9, 0x01, 0x22,    // switch precharge to 0x22 // 0xF1
    0xDA, 0x01, 0x12,    // set divide ratio
    0xDB, 0x01, 0x20,    // vcom deselect to 0x20 // 0x40
    0x8D, 0x01, 0x14,    // Enable charge pump
    0xA4, 0x00,          // Display resume
    0xAF, 0x00,          // Display on
};
*/
//ROTATED 180 DEGREES
static const PROGMEM uint8_t s_SSD1306_lcd128x64_initData[] =
{
#ifdef SDL_EMULATION
    SDL_LCD_SSD1306, 0x00,
    0x00, 0x00,
#endif
    0xAE, 0x00,          // display off
    0x20, 0x01, 0x00,    // Page horizontal Addressing mode
    0xC0, 0x00,          // Scan from 127 to 0 (Reverse scan)
    0x40| 0x00, 0x00,    // First line to start scanning from
    0x81, 0x01, 0x7F,    // contast value to 0x7F according to datasheet
    0xA0| 0x00, 0x00,    // Use reverse mapping. 0x00 - is normal mapping
    0xA6, 0x00,          // Normal display
    0xA8, 0x01, 63,      // Reset to default MUX. See datasheet
    0xD3, 0x01, 0x00,    // no offset
    0xD5, 0x01, 0x80,    // set to default ratio/osc frequency
    0xD9, 0x01, 0x22,    // switch precharge to 0x22 // 0xF1
    0xDA, 0x01, 0x12,    // set divide ratio
    0xDB, 0x01, 0x20,    // vcom deselect to 0x20 // 0x40
    0x8D, 0x01, 0x14,    // Enable charge pump
    0xA4, 0x00,          // Display resume
    0xAF, 0x00,          // Display on
};
`

lexus2k, 2021-02-07:
Hi,

You don't need to change initialization code. You have to use `display.getInterface().flipHorizonal()` and `display.getInterface().flipVertical()` methods.

jancotipsarevic, 2021-02-07:
Hello, well what you mention was one of my first tries but it looks like none of this methods appears under display, 
This is the error:

main.cpp:45:28: error: ‘class InterfaceSSD1306<PlatformI2c>’ has no member named ‘flipHorizonal’; did you mean ‘flipHorizontal’?
     display.getInterface().flipHorizonal();
                            ^~~~~~~~~~~~~
                            flipHorizontal
Process terminated with status 1 (0 minute(s), 1 second(s))
1 error(s), 0 warning(s) (0 minute(s), 1 second(s))
 
And this is my code:

#include <cstdlib>
#include <iostream>
#include <chrono>
#include <thread>
#include "lcdgfx.h"

using namespace std;

int main(int argc, char **argv)
{

    DisplaySSD1306_128x64_I2C display((int8_t) 5, { -1, 0x3D, -1, -1, 0 });
    display.getInterface().flipHorizonal();
    display.begin();
    display.clear();
    display.drawLine(10,10,50,10);
    display.setFixedFont(ssd1306xled_font6x8);
    display.printFixed (0,  8, "Line 1. Normal text", STYLE_NORMAL);

     cout << "Finished!, press any key to continue" << endl;
     cin.get();

     return EXIT_SUCCESS;
}

Chrispren, 2021-02-07:
I think you've misspelled "flipHorizontal" You have "flipHorizonal"

jancotipsarevic, 2021-02-07:
Sorry, in this second try I've mispelled, now corrected but the issue persists as I saw previously.
Here again the error and the code.
//Error 
/home/pi/projects/gpiod_examples/oled_use_lib/main.cpp|45|error: no matching function for call to ‘InterfaceSSD1306<PlatformI2c>::flipHorizontal()’|

//Code
#include <cstdlib>
#include <iostream>
#include <chrono>
#include <thread>
#include "lcdgfx.h"

using namespace std;

int main(int argc, char **argv)
{

    DisplaySSD1306_128x64_I2C display((int8_t) 5, { -1, 0x3D, -1, -1, 0 });
    display.getInterface().flipHorizontal();
    display.begin();
    display.clear();
    display.drawLine(10,10,50,10);
    display.setFixedFont(ssd1306xled_font6x8);
    display.printFixed (0,  8, "Line 1. Normal text", STYLE_NORMAL);

 	cout << "Finished!, press any key to continue" << endl;
 	cin.get();

	return EXIT_SUCCESS;
}

Chrispren, 2021-02-07:
Try this?

int main(int argc, char **argv)
{
DisplaySSD1306_128x64_I2C display((int8_t) 5, { -1, 0x3D, -1, -1, 0 });
display.begin();
display.clear();
display.getInterface().flipHorizontal(1);
display.drawLine(10,10,50,10);
display.setFixedFont(ssd1306xled_font6x8);
display.printFixed (0,  8, "Line 1. Normal text", STYLE_NORMAL);

// cout << "Finished!, press any key to continue" << endl;
// cin.get();

return EXIT_SUCCESS;
}

jancotipsarevic, 2021-02-07:
EXCELLENT!! now works, now rotated 180 degrees!.
Congratulations this is a great library!!!!

Thank you very much.

The code now is:

int main(int argc, char **argv)
{

    DisplaySSD1306_128x64_I2C display((int8_t) 5, { -1, 0x3D, -1, -1, 0 });
    display.begin();
    display.clear();
    display.getInterface().flipHorizontal(1);
    display.getInterface().flipVertical(1);
    display.drawLine(10,10,50,10);
    display.setFixedFont(ssd1306xled_font6x8);
    display.printFixed (0,  8, "Line 1. Normal text", STYLE_NORMAL);

 	//cout << "Finished!, press any key to continue" << endl;
 	//cin.get();

	return EXIT_SUCCESS;
}

closed 2021-02-07 by jancotipsarevic

Chrispren, 2021-02-07:
Glad you got it running :-)

lexus2k, 2021-02-07:
@jancotipsarevic  I', sorry for misspell. The comment is fixed.
@Chrispren  Thank you for clarifications