lexus2k/lcdgfx #44: OLED 0,96 - problem with displaying a variable

szymonsmolka, 2021-02-11:
Ask any question, you have, regarding the library
Hi. I'm trying to use the lcdgfx library to run a 0.96 64x128 OLED display with the SH1107 driver. I wanted to make a simple thermometer. The temperature is assigned to the float variable, but the display does not show it.
display.printFixed (0, 0, temp, STYLE_BOLD);

How do I enter such a code for example
display.printFixed (0, 0, "TEST", STYLE_BOLD);

The display works and shows TEST

Where is the problem? Please help.

lexus2k, 2021-02-11:
Please, try:

```.cpp
    display.setTextCursor(0,0);
    display.print( temp );
```

The function supports only int variables

szymonsmolka, 2021-02-12:
The temperature is displayed. Thank you. However, there is no decimal value. Is there a way to do this?
Can the font size be changed in this function?

lexus2k, 2021-02-16:
@szymonsmolka 

I'm sorry for the delay. Actually you can display any numbers you want like this:
```.cpp
   char str[10];
   sprintf( str, "%f.2", temp );
   display.print( str );
```

szymonsmolka, 2021-02-17:
No problem :) But it doesn't help. The value of 1266 is displayed all the time. Fortunately, I managed differently with decimal values.
Unfortunately, I can't enlarge the font. Is it possible at all?

szymonsmolka, 2021-02-19:
> ```c++
> char str[10];
>    sprintf( str, "%f.2", temp );
>    display.print( str );
> ```

Have an idea why the value 1266 is displayed?

lexus2k, 2021-02-19:
Hi, I'm sorry, that was the wrong location of `f` character in format string.
I fixed the format string in the examples above.

Anyway, there is very rich documentation on `printf` and `sprintf` functions over internet. Please refer to documentation.

lexus2k, 2021-02-19:
[image]

lexus2k, 2021-02-19:
> Can the font size be changed in this function?

No, the only supported feature is for monochrome displays. `printFixedN`

szymonsmolka, 2021-02-20:
> Anyway, there is very rich documentation on `printf` and `sprintf` functions over internet. Please refer to documentation.

After your answer, I did so. The code looks fine, and so on it returns 1266. For example, if I change str [10] to str [15], the value is 1249. Can you take a look at what's wrong with the code?

`#include "lcdgfx.h"
#include "lcdgfx_gui.h"
#include <Arduino.h>
#include <OneWire.h>
#include <DallasTemperature.h>
#include <Wire.h>

OneWire oneWire(A0);
DallasTemperature sensors(&oneWire);

DisplaySH1107_64x128_I2C display(-1);

float temp;

void setup()
{
  display.begin();
  display.setFixedFont(ssd1306xled_font6x8);
  display.clear();
  display.fill(0x00);
  sensors.begin();
}

void loop()
{

  sensors.requestTemperatures();
  temp = sensors.getTempCByIndex(0);

  display.setTextCursor(0, 64);
  char str[10];
  sprintf( str, "%.2f", temp );
  display.print( str );

  delay (5000);

}`

lexus2k, 2021-02-20:
Which version of Arduino IDE are you using? Does it display any warnings / errors.

Try this code:
```.cpp
char str[10];
sprintf( str, "%.2f", temp );
display.write( str );
```

szymonsmolka, 2021-02-20:
> Which version of Arduino IDE are you using?

Arduino IDE 1.8.13

> Does it display any warnings / errors?

Summary after upload:

Sketch uses 10044 bytes (70%) of program storage space. Maximum is 14336 bytes.
Global variables use 336 bytes (32%) of dynamic memory, leaving 688 bytes for local variables. Maximum is 1024 bytes.

> ```c++
> char str[10];
> sprintf( str, "%.2f", temp );
> display.write( str );
> ```

After this change, the display only shows a question mark :(
Also no warnings.

lexus2k, 2021-02-20:
I don't have all those sensors. so I commented out the code reading the temperature, and replaced it with single assignment. And it works. So the issue somewhere in another library, and not in lcdgfx

```.cpp
#include "lcdgfx.h"
#include "lcdgfx_gui.h"
//#include <Arduino.h>
//#include <OneWire.h>
//#include <DallasTemperature.h>
//#include <Wire.h>

//OneWire oneWire(A0);
//DallasTemperature sensors(&oneWire);

DisplaySH1107_64x128_I2C display(-1);

float temp;

void setup()
{
  display.begin();
  display.setFixedFont(ssd1306xled_font6x8);
  display.clear();
  display.fill(0x00);
  // sensors.begin();
}

void loop()
{

  // sensors.requestTemperatures();
  temp = 23.23; // sensors.getTempCByIndex(0);

  display.setTextCursor(0, 64);
  char str[10];
  sprintf( str, "%.2f", temp );
  display.write( str );

  delay (5000);

}
```

The result:
[image]

szymonsmolka, 2021-02-21:
An interesting thing. When I do what you did, it still displayed a question mark. Maybe my display is broken.

lexus2k, 2021-02-23:
@szymonsmolka What about direct_draw/draw_text example? Can you capture the result for that example for me?

szymonsmolka, 2021-02-24:
Of course I can, but which is an example?

szymonsmolka, 2021-02-25:
> @szymonsmolka What about direct_draw/draw_text example? Can you capture the result for that example for me?

@lexus2k 

[https://github.com/lexus2k/lcdgfx/blob/master/examples/direct_draw/draw_text/draw_text.ino](url)

not working :(

closed 2021-03-09 by szymonsmolka

Mephisto68, 2023-10-25:
> > Can the font size be changed in this function?
> 
> No, the only supported feature is for monochrome displays. `printFixedN`

Hello,
We cannot "send" this to the screen:
Display.printFixedN(10, 52, var, STYLE_NORMAL, FONT_SIZE_2X);
But what a shame not to be able to do that!
I will have to get another library
However, yours is so good!
Would it not be possible in a future version to add this function?

Mephisto68, 2023-10-25:
include "lcdgfx.h"
#include "lcdgfx_gui.h"

/* LCD */
  #define PORT_ECRAN_RESET      10
  #define PORT_ECRAN_CS         8
  #define PORT_ECRAN_DC         9
  #define PORT_ECRAN_MOSI_SDA   11
  #define PORT_ECRAN_SCK_SCL    13
  #define PORT_ECRAN_ECLAIRAGE  7
  DisplayST7735_128x128x16_SPI display = DisplayST7735_128x128x16_SPI(PORT_ECRAN_RESET,{-1, PORT_ECRAN_CS, PORT_ECRAN_DC, 0, PORT_ECRAN_MOSI_SDA, PORT_ECRAN_SCK_SCL});

float temp;

void setup()
{
  display.begin();
  display.setFixedFont(ssd1306xled_font6x8);
  display.clear();
  display.fill(0x00);
  // sensors.begin();
}

void loop()
{

  // sensors.requestTemperatures();
  temp = 23.23; // sensors.getTempCByIndex(0);

  display.setTextCursor(20, 20);
  char str[10];
  sprintf( str, "%.2f", temp );
  display.write( str );

  delay (5000);

}
> RESULT  :  the display only shows a question mark :(

lexus2k, 2023-10-29:
Hi, 

I apologize for this inconvenience. However the library has some limitations to reduce flash ram usage.
New functionality can be implemented, it requires some time.

Best regards,
Aleksei