lexus2k/lcdgfx #45: How to use DisplaySSD1306_128x32_CustomI2C

dansleboby, 2021-02-13:
Hi I don't have access to standard I2C pin how I can use a custom one?

Thanks

lexus2k, 2021-02-13:
Hi,
There is no ready to use example for custom I2c, but there is an example [CustomSPI](https://github.com/lexus2k/lcdgfx/blob/master/examples/interfaces/custom_spi/custom_spi.ino).
You can use it as base for i2c development

Best regards

lexus2k, 2021-02-20:
Hi, did you figure out how to use CustomI2C?

> I don't have access to standard I2C pin how I can use a custom one?

if you still have the problem, you need to provide more information on hardware platform, schematics of your setup, the code example. Not all pins can be used for i2c communication.

Best regards

closed 2021-03-29 by lexus2k