lexus2k/lcdgfx #46: how to rotate the screen with SSD1306/I2C

pkerspe, 2021-02-21:
Sweet little library and especially the speed oposed to overloaded ones like the adafruit one.

Just wondering how one could rotate the screen (flip vertically and horizontally) using this library?
Is there any documentation of all exposed functions once can use on the display instance?
This library would be perfect if there was some more documentation available on the different functions

lexus2k, 2021-02-22:
Hi Paul,

There is no way rotate by 90 degrees. But you can flip horizontally or vertically:
```.cpp
display.getInterface().flipVertical( 0 or 1 );
display.getInterface().flipHorizontal( 0 or 1 )
```

Best regards.

lexus2k, 2021-02-22:
> This library would be perfect if there was some more documentation available on the different functions

documentation is automatically generated from the code by doxygen. Unfortunately, I don't have much time to write much docs, but examples can be a good start point to understand the library.

Here is a [link to the functions in the documentation](https://codedocs.xyz/lexus2k/lcdgfx/class_interface_s_s_d1306.html#aceb904ec0ca42dfce682a2dac226a404), you're looking for

pkerspe, 2021-02-22:
Thanks a lot Aleksei for the quick response.
The hint about using "getInterface" was my missing link :-) I noticed the two functions in the interface but did not know how to actually get the access to interface.
Problem solved now, thanks!
One example showing this function would probably help future users. I did not find an example for this at least, bit now I know it anyway :-)

lexus2k, 2021-02-22:
I agree.

The idea of the library architecture: all common functions for all displays are available via `display.` object. While some displays have hardware acceleration or specific functions. Those ones are available via `display.getInterface().` object