lexus2k/lcdgfx #47: Set different sizes on one Font

Danixu, 2021-02-24:
Hello,

I'm migrating from Adafruit library to this library to control an ST7735 display, because uses less RAM and Program space and my program need it.

My question is if is possible to use one font and change their size, because If not I will have to use two different fonts and uses more program memory. I've found that exists a function called printFixedN, but or I don't know how to use it, or is not available to my display.

Thanks!

Danixu, 2021-02-24:
Looking into another issue I have seen this:

> > Can the font size be changed in this function?
> 
> No, the only supported feature is for monochrome displays. `printFixedN`

So I think that maybe my question is answered...

lexus2k, 2021-02-24:
HI @Danixu,

Unfortunately printFixedN works only for monochrome displays. I can try to do quick porting of this function for color displays, but the speed of printFixedN will not be as fast, as it could.

Danixu, 2021-02-24:
Thanks!!, I will try to figure out which letters are being used on my project to remove the others and try to free some space. Without do it I have already saved about 14% of Program Memory just changing the library, so maybe is enough for my program.

lexus2k, 2021-02-24:
Please, check latest commit with printFixedN support for 16-bit displays

Danixu, 2021-02-25:
Oh, thanks!!. I'll take a look

Best regards.

closed 2021-02-25 by Danixu