lexus2k/lcdgfx #48: SH1107 128x64 offset and sideways

spotrevoc, 2021-02-24:
Using ESP32 with Adafruit OLED Featherwing display (SH1107 128x64 I2C). 

lcdgfx library version 1.0.6

When running the sh1107_demo sketch from the example folder, image is rotated 90 degrees and only works in bottom right 64x32 pixels.  Bottom left 64x32 pixels show a snowy pattern. 

64x128 example works fine but screen is rotated and does not fit my needs that way.

Compiled using Arduino IDE 1.8.13 in Windows 10

[image]

lexus2k, 2021-02-24:
@spotrevoc 
hi

Are you sure you have 128x64 display, and not 64x128?

spotrevoc, 2021-02-24:
I believe technically it's a 64x128. Adafruit markets it as 128x64 but as I'm diving into their library they document it as "rotated" 64x128.

spotrevoc, 2021-02-25:
Also, nothing against Ada, but I really like your library overall and would like to stick with it since I've already developed a lot of code with it for an SH1106.  I had hoped the Adafruit SH1107 oled shield would be pretty much a drop in replacement.

lexus2k, 2021-02-25:
Adafruit SH1107 is a good library, like many others Adafruit libraries. But they use a lot of virtual functions in their implementation, and their library consumes too much flash size and memory. And profit they have is ease of new features development. And they always use double buffering which eats 1024 bytes of RAM in case 128x64 monochrome displays.
So, to support specific display, or rotation feature they need to rewrite only single method.

lcdgfx, unlike Adafruit library, is built around C++ templates. It provides similar interface for users, but implementation is much more difficult. The advantage of such approach  is small flash, ram requirements and faster speed.  lcdgfx supports  direct draw functions (to draw on the display without double buffering), and nano engine - use double buffering to refresh only part of the display content. 
Some specific displays, rotation modes in lcdgfx library require too much code. That's not easy.

Actually there are 2 types of 128x64 lcd displays based on SH1107 controller. And there orientation is hardware feature (you can see the difference in flex cable connection to the matrix):
[image]
[image]

> How easy could it be to swap x/y and add a separate constructor with the same startline and offset they use?

The offset has nothing to do with the problem solution. And it is already used by lcdgfx: `"    0xD3, 0x01, 0x60,  // offset 0x60 according to datasheet",`

> If that's not feasible, I really don't mind using it as 64x128 and swapping cursor position x and y in my code, as long as there's a way to rotate text and sprites with the library.

That's not easy. The most simple solution in your case is to use canvas and write new method `rotate` for it.

lexus2k, 2021-03-29:
I'm closing the issue, since there is no any activity. But feel free to  reopen if you have any questions

closed 2021-03-29 by lexus2k

guhland, 2021-11-18:
How would I go about using canvas to write a rotate method?

Pedrohpcavalcante, 2022-02-20:
How would I go about using canvas to write a rotate method?

lexus2k, 2022-03-09:
The solution for 1-bit displays is in #72 ticket