lexus2k/lcdgfx #49: Definition of pins for the SPI

mkhornet, 2021-03-04:
Hi! I am trying to use this lib with my new display over SPI. Somehow after digging through the examples or documentation I cannot understand the how are the hardware pins defined (in this call:
`DisplaySH1106_128x64_SPI display(3,{-1, 4, 5, 0,-1,-1});   // Use this line for Atmega328p (3=RST, 4=CE, 5=D/C)`
I assume there are pin numbers in here but what is the layout of this command (the documentation is unclear for me about this matter)
BR

lexus2k, 2021-03-05:
https://codedocs.xyz/lexus2k/lcdgfx/class_display_s_h1106__128x64___s_p_i.html#aa71de050d7ed722fea8a5b9b3f48e352

https://codedocs.xyz/lexus2k/lcdgfx/struct_s_platform_spi_config.html

mkhornet, 2021-03-05:
Yes, I have found those but that is exactly what i don't understand. You write there as if there are only RST DC and CE pins needed for displaying when using something like arduino. I leave MOSI and CS pins open but I can't get it to work on 3 pins only. I have tested it with other libs (adafruit for SSD1306) and the screen in working.

lexus2k, 2021-03-06:
All supported displays are verified. So particular the problem is in your specific case. You need to provide much more details: platform, ide being used, hardware schematics, code example, etc.
Next, you're saying "adafruit for ssd1306", and the constructor is for sh1106.
The description of the problem is very confusing.

mkhornet, 2021-03-06:
Sorry if I am confusing you. I have mentioned the ssd1306 only because that was the first lib i tested (with adafruit lib) that confirmed my screen works. The screen I am using is using the sh1106 (at least that is what the manufacturers page says). The platform i am using is arduino nano clone and IDE is arduino IDE 1.8.13. The code I am using is one of examples loaded with your library. for HW connections: D3 ->RST D4->CLK and D5 ->DC

mkhornet, 2021-03-06:
Ok, I'm a potato. Got this solved, I missunderstood the pin assignment. Thanks for the support anyways!

closed 2021-03-06 by mkhornet