lexus2k/lcdgfx #50: GUI Menu and sub-menus

lquang4321, 2021-03-08:
Before I proceed, I'm a beginner in terms of programming so I might used incorrect programming terms. Anyways, I'm making a little device that needs a simple UI library. I managed to create the main menu but it's difficult to create submenu. At first, I tried to create a multidimensional array and then cycle through the "pages" but I don't think the menu gets updated. I'm at a lost. My current solution is to create multiple LcdGfxMenu object and manually do the logic so it displays ontop which gets tedious as the submenu grows.
``` cpp
const char *menuPG[][4] =
{
    { "Main 1", "Main 2", "" ,  ""},                   //Main menu
    { "Sub 1", "Sub 2", "Sub 3", "Back"},        //Submenu Page 1
    { "Sub 1", "Sub 2", "Sub 3", "Back"},       //Submenu Page 2
};
uint8_t curPAGE= 0;
SAppMenu menu;

void setup()
{
    display.begin();
    display.clear();
    display.setFixedFont(ssd1306xled_font6x8);
    display.createMenu(&menu, menuPG[curPAGE], sizeof(menuPG[curPAGE]) / sizeof(char *), {0,9} );  

   /* The only way for a submenu to work is manually calling each submenu below to override the previous */                               
    //display.createMenu(&set_up, set_upPG, sizeof(set_upPG) / sizeof(char *) );
    //display.createMenu(&sensors, sensorPG, sizeof(sensorPG) / sizeof(char *), {82, 36, 127, 64} );
    //display.createMenu(&about, aboutPG, sizeof(aboutPG) / sizeof(char *) );
    display.showMenu(&menu);
}

void loop()
{
    curPAGE ++;
    display.updateMenu(&menu);
    display.showMenu(&menu);
    delay(1000);
}
```

Is there a way to create a submenu system?  I'm currently using Adafruit nrf52 Feather bluefruit and SSD 1306 0.96" mono. Thanks.

lexus2k, 2021-03-09:
Hi,

Unfortunately, submenu functionality is not implemented.  You're thinking in the correct direction:
`createMenu(SAppMenu *menu, const char **items, uint8_t count, const NanoRect &rect)`
This method is the right way to implement submenus. You're only need to remember which level is active now, something like this (pseudocode):

```.cpp
SAppMenu *activeMenu = &mainMenu;

void loop()
{
    ...
    display.showMenu( activeMenu );
    ...
    if item is selected then activeMenu = &subMenu;
}
```

Best regards.

lquang4321, 2021-03-13:
Thank you for your help, but since this is a school project I can't waste time. I decided it's easier to use ArduinoMenu library although that gave me headache of its own due to the shear amount of features/functionality.

closed 2021-03-13 by lquang4321