lexus2k/lcdgfx #51: Support for ST7735 80x160 0,96''

T94T, 2021-03-13:
Is there a way to support the smaller ST7735 TFT display?
[image]

lexus2k, 2021-03-13:
Hello again,

So, could you please send me a link to datasheet, and let me know what happens, when using DisplayST7735_128x160x16_CustomSPI constructor. Thank you

Best regards

T94T, 2021-03-13:
It's working quite well at this point. I think, it is only a question of resolution. 
[0.96 inch 80x160 IPS LCD.pdf](https://github.com/lexus2k/lcdgfx/files/6134934/0.96.inch.80x160.IPS.LCD.pdf)

[image]

lexus2k, 2021-03-14:
Please, find initial support for the display on master branch.
For testing, don't forget to change 128x160 in the demo code to 80x160 (there are 2 such places).
And could you please capture screenshots for me for all rotation options (0,1,2,3)?

PS. Don't worry about `drawBitmap()` issues with owl. That's because the picture is of 128x64 size, while drawBitmap() doesn't support clipping.

T94T, 2021-03-15:
These are the results:
**Rotation 0**
[image]
**Rotation 1**
[image]
**Rotation 2**
[image]
**Rotation 3**
[image]

lexus2k, 2021-03-15:
Thank you. It looks like real lcd works, the problem only remains in the emulator.

closed 2021-03-29 by lexus2k