lexus2k/lcdgfx #52: Render of space with free font

gpfreund, 2021-03-17:
Thanks for writing and maintaining this great library - much appreciated. I did run into one problem: I generated a font from OpenSans and used printFixed() to display some text. However and space (0x20) character renders as previous character in the string. For example "Bold text" renders as "Bolddtext".

### 1) Generating the font
`.\fontgenerator.py --ttf opensans.ttf -s 16 -g 0 127 -f new -d > output.cpp`

### 2) Initializing the OLED
`DisplaySSD1327_128x128_I2C display(-1);`

### 3) Render the text (based on the example code)
 
```static void textDemo()
{
    display.setFreeFont(free_opensans24x22);
    display.clear();
    display.setColor(GRAY_COLOR4(255));
    display.printFixed(0,  8, "Normal text", STYLE_NORMAL);
    display.setColor(GRAY_COLOR4(192));
    display.printFixed(0, 34, "Bold text", STYLE_BOLD);
    display.setColor(GRAY_COLOR4(128));
    display.printFixed(0, 60, "Italic text", STYLE_ITALIC);
    display.setColor(GRAY_COLOR4(164));
    display.invertColors();
    display.printFixed(0, 86, "Inverted bold", STYLE_BOLD);
    display.invertColors();
    delay(20000);
}
```

### Result:
[image]

lexus2k, 2021-03-17:
@gpfreund Hello,

Can you please share opensans.ttf file? I don't have that one.

lexus2k, 2021-03-17:
It seems, I know what is the root cause. 
Please don't use non-printable chars ascii codes in the generator.
Instead of `.\fontgenerator.py --ttf opensans.ttf -s 16 -g 0 127 -f new -d > output.cpp`, please use `.\fontgenerator.py --ttf opensans.ttf -s 16 -g 32 127 -f new -d > output.cpp`
I've also did the fix to the font generator to validate chars, when generating output c-file.

gpfreund, 2021-03-19:
Hi, thanks for the quick response. The Open Sans file is attached (it's really Open Sans Normal). The whole series is here: [Open Sans
](https://fonts.google.com/specimen/Open+Sans?preview.text_type=custom). 
[opensans.zip](https://github.com/lexus2k/lcdgfx/files/6168404/opensans.zip)

The issue itself is fixed - in record time! Very much appreciated! The only other thing I noted is that there when inverting the text, there is a single black pixel separating the characters and it only inverts the extend of the actual character. I would expect that it inverts the maximum height of the character (including the area below the base line). Not a big deal for us but just FYI.
[image]

closed 2021-03-19 by gpfreund