lexus2k/lcdgfx #56: GPIO input on RaspberryPi

m3astwood, 2021-04-25:
Ask any question, you have, regarding the library
Hello, 
I'm quite new to C++ and my problem is probably user error. I am unable to read pin inputs from the GPIO while able to write without issues. I have managed to get my functionality working with WiringPi (to test my circuit) but am unable to translate it to use the lcd_gpioRead() function. 

All I'm trying to do is get the input of a push button. GPIO and WiringPi pin numbers are different for the same pins. Thanks in advance for any help...

My WiringPi code :
```c++
#include <stdio.h>
#include <wiringPi.h>

int main() {
  wiringPiSetup();

  pinMode(22, INPUT);

  while (true) {
    printf("digital read %d\n", digitalRead(22));
  }

  return 0;
}
```

My lcdgfx code :
```c++
#include <stdio.h>
#include "lcdgfx.h"

int main () {
  lcd_gpioMode(6, LCD_GPIO_INPUT);

  while (true) {
    printf("gpio read %d\n", lcd_gpioRead(6));
    // testing adcRead()
    // printf("adc read %d\n", lcd_adcRead(6));
  }

  return 0;
}
```

lexus2k, 2021-04-25:
Hello,

Does it show in the console anything like this: "Failed to set gpio pin direction2", "Failed to set gpio pin direction1", "Failed to allocate gpio pin" etc?
What is the content of `/sys/class/gpio/gpio6/direction` file, when the program running?
I'm sorry, I don't have raspberry by hand right now.

WirePi works through BCM chip library, while lcdgfx uses the standard linux interface GPIO class.

m3astwood, 2021-04-26:
Thanks for responding so quickly. I don't get any errors from the console, and when I check the `/sys/class/gpio/gpio6/direction` it does read _in_. 

I tried pressing my button and checking the `/sys/class/gpio/gpio6/value` and it does in fact change from 1 to 0.

lexus2k, 2021-04-26:
@m3astwood 

I'm so sorry for that. GPIO read is not implemented for linux now

```.cpp
int lcd_gpioRead(int pin)
{
    return LCD_LOW;
}
```

But you can quickly replace the code in the file lcdgfx/src/lcd_hal/linux/platform.cpp at lines 310 - 313:
```.cpp
int lcd_gpioRead(int pin)
{
    return gpio_read(pin) ? LCD_HIGH : LCD_LOW;
}
```

Let me know if it works

m3astwood, 2021-04-27:
Cool thanks, I'll give that a go and report back.

m3astwood, 2021-04-27:
@lexus2k that worked a charm. Thanks so much for your help!

lexus2k, 2021-05-07:
Hi, I added support for Linux to read gpio.
Let me know if the fix fits your needs.

Thank you

lexus2k, 2021-05-13:
Feel free to reopen the issue, if the issue still exists.

closed 2021-05-13 by lexus2k

m3astwood, 2021-05-13:
Thanks - yes I've had to put my project on hold for a little so I haven't been able to verify yet but when it comes back to the top of my list I'll update you.

m3astwood, 2021-05-21:
@lexus2k can confirm it is working great! thanks for you help :+1: