lexus2k/lcdgfx #57: how to print data from sensors?

Indra-nil, 2021-05-07:
hi,
recently i'm using lcdgfx library due to low resource consumption.
the only problem is that i'm unable to print raw and calculated float data from sensors to LCD.
i'm using,
display.printFixedN (XX, XX, "XX", STYLE_NORMAL, FONT_SIZE_4X); 

but this command only supports string value.

lexus2k, 2021-05-10:
Please, start from reading #42

Indra-nil, 2021-06-21:
thanx for the suggestion that's working.
I also tried #44 example u gave but that didn't work at all
only showing "?".

Indra-nil, 2021-06-21:
"Due to some performance reasons %f is not included in the Arduino's implementation of sprintf()"
instead of sprintf() one should use dtostrf()
"dtostrf(temp, 3, 2, str);"
some links : [(https://stackoverflow.com/questions/27651012/arduino-sprintf-float-not-formatting)]
[(https://arduino.stackexchange.com/questions/26832/how-do-i-convert-a-float-into-char)]

thank you by the way.

lexus2k, 2021-06-22:
> Due to some performance reasons %f is not included in the Arduino's implementation of sprintf()

Oh, my bad. I'm sorry to mislead. Well, I will prepare the fix soon.

closed 2021-06-22 by Indra-nil