lexus2k/lcdgfx #60: Cannot build using this library for esp32

blmhemu, 2021-05-23:
**Describe the bug**
Cannot build using the library.

**To Reproduce**
1. Go to esp-arduino-blink example in platformio
2. Add this library as dependency
3. use `#include "lcdgfx.h"` in the main.cpp
4. Build the project.

**Expected behavior**
Program should build

**Screenshots**
[image]

**Please complete the following information:**
 - 1.1
 - SSD1306
 - MacOS
 - ESP32
 - VSCODE - PlatformIO

**Additional context**
None

lexus2k, 2021-05-23:
Hi,

Can you try to add -DARDUINO to your project defines?

blmhemu, 2021-05-24:
Thanks for the quick response!

[image]
After I added `-D ARDUINO` I got the above error

Actually I am using arduino AND esp-idf (Both of them as the project is dependent on some arduino libs)

lexus2k, 2021-05-24:
Please, find the fix on the master branch (latest commit), ARDUINO define is not required.

blmhemu, 2021-05-25:
Greetings ! @lexus2k 
That was a nice fix. The problem can be marked as resolved In master. Eager to see it in new release !

Once again thanks for the quick fix !

closed 2021-05-25 by blmhemu