lexus2k/lcdgfx #62: Question about png

emanueleci, 2021-06-10:
Hi,
i have an OLED SSD1351 SPI 128*128 Color screen and i simple need to get (wget), resize and show a png or jpeg image on the screen. 
I can't find any clear documentation or question about the possibility to use your lib on raspberry pi (c++) for that goal.
Please, can you tell me if your lib expose some api to resize and display simple color images?

Thank you!
Emanuele

lexus2k, 2021-06-10:
Hi Emauele,

In the `tools` folder there is a script, which allows to compile any example for Linux-based platform. If you have SDL2 installed, you can even run on PC with a Linux:

> ./build_and_run.sh -p linux -e -f demos/ssd1351_demo

If you want to compile and run on RPi, then just remove `-e` option (which means to use SDL2):

> ./build_and_run.sh -p linux -f demos/ssd1351_demo

The line above should run demo on RPi with SSD1351 display connected. Please, don't forget to change in the ssd1351 demo code constructor to RPi. Also, remember that the library works from user space and requires spidev linux kernel module to be loaded properly.

FYI, Also Linux has FrameBuffer driver for the oled display, and you also can use standard Linux OS capabilities to draw on the display (but that can be complicated)

emanueleci, 2021-06-11:
Ok, thank you for your time.
FIrst step is trying to run demo on **Raspberry PI4:**
Modified demo 1351.ino to use Rasberry and selected this: DisplaySSD1351_128x128x16_SPI display(24,{-1, 0, 23, 0,-1,-1});
Compiled ok, demo starts ok in console but screen stay completly black.
So tried to change ping configuration based on my setup:
(this is the pin configuration used for a Python lib. This configuration works with my display on this system:
Clock 23
MOSI 19
Reset 22
Data/Command 18
Chip Select 24

so based on documentation, i changed constructor to:
DisplaySSD1351_128x128x16_SPI display(22,{24, -1, 18, 0, 23, 19});

but receive SPI error:
Failed to initialize SPI: No such file or directory!
SPI failed to send SPI message: Bad file descriptor

But i think SPI is correctly loaded in my system;
ls -l /dev/spi*
crw-rw---- 1 root spi 153, 0 giu 10 22:18 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 giu 10 22:18 /dev/spidev0.1

and 

lsmod | grep spi_
spi_bcm2835            20480  0

maybe i misunderstood constructor pin assignemnt?
https://codedocs.xyz/lexus2k/lcdgfx/class_display_s_s_d1351__128x128x16___s_p_i.html
https://codedocs.xyz/lexus2k/lcdgfx/struct_s_platform_spi_config.html

Thank you

emanueleci, 2021-06-14:
After many attemps, i was not able to make demo works on RPI4 for ssd1351. Maybe example is not ready for that system, or my pins configuration are wrong.

lexus2k, 2021-06-14:
All examples are tested on Rasperry Pi and work.
I'm sure that something is wrong either with pin configuration/connections or linux drivers in your case.

emanueleci, 2021-06-15:
> All examples are tested on Rasperry Pi and work.
> I'm sure that something is wrong either with pin configuration/connections or linux drivers in your case.

Can you gently tell me the pin configuration I should use and the constructor parameters? I can change my configuration without problem.
Thank you

emanueleci, 2021-07-06:
Hi @lexus2k,
after changed my pins configuration, i was able to compile and run your demo.
This is the working connection for me:
DIN Pin 19 (GPIO10)
CLK Pin 23 (GPIO11)
Cd Pin 24 (GPIO8)
DC Pin 16 (GPIO23)
RST Pin 18 (GPIO24)
-> DisplaySSD1351_128x128x16_SPI display(24,{-1, {0}, 23, 0,-1,-1});  

Now i'm trying to show an image in the oled screen.
The format of the image i need to show is **JPEG** (not PNG as I had thought). 
I suppose i need some lib to:

- Decoder Jpeg
- resize image at screen size (128*128)
- Transform output to some usefull for your lib drawBitmap (what format? 8,16,24 bit?)

do you have any suggestions for C++ lib?

Thank you!
Emanuele

lexus2k, 2021-07-06:
Hi @emanueleci ,

I'm sorry that I didn't answer you yet regarding raspberry pi pins. I have one display connected via SPI to raspberry.
Regarding JPEG decoding library, that one can be useful for you (but I didn't try it): https://github.com/bitbank2/JPEGDEC
lcdgfx library currently supports only 16-bit displays in 5-6-5 mode. Thus, if you need the transformation from RGB 24-bit, you need to do bit-wise  procedure for each pixel.

You can use the macro:
```.cpp
#define RGB_COLOR8(r, g, b) ((r & 0xE0) | ((g >> 3) & 0x1C) | (b >> 6))
```

closed 2022-02-14 by lexus2k