lexus2k/lcdgfx #63: Scroll text

tomudo, 2021-06-29:
**Scroll text feature.**
I tried to write code for scroll text by use  printFixed( x_pos, y_pos, "text", any_style); but when I scroll to right by increase x_pos beyond edge of screen. The text is mess up by jumping to next line. I want a solution to scroll text to right or left for text longer than screen.

lexus2k, 2021-06-29:
Hi.

Can you provide the code to demonstrate the issue?

tomudo, 2021-06-30:
My code looks like this.

void scrollText(){
 for(int i=0;i<160;i++){
  display.printFixed(i ,  32, "Hello world", STYLE_NORMAL);
 }
}

I will look into using canvas technic. Thank you.

lexus2k, 2021-06-30:
If you're not limited in memory, you can create `NanoCanvas<128,64,1>`. It needs only 1024 bytes of RAM.
Please make sure that canvas property `CANVAS_TEXT_WRAP_LOCAL` is disabled.