lexus2k/lcdgfx #66: LCD font generator app for type 0 fonts

Camarine11, 2021-07-17:
Hi

I have question about what app for LCD font generator  can be used to generate new type 0 fonts.
Can I use the free LCD font generator from MikroC? If this app is not good for this what app or online LCD font generator to use?

I need to make a 51 or 55 pixel height font with low spacing (eliminate all vertical and horizontal lines with white pixels) to fit 5 characters on a 1.3 inch SH1106 128x64 oled screen. 
LCD font generator from MikroC is nice app that shows the font image but did not work for me.

Found next in font_format.txt
    - type 0 fonts support only ascii chars with codes 0 - 127, and extended chars with codes 128 - 255.
      These fonts can be generated with LCD font generator ONLY.

Thank you

lexus2k, 2021-07-18:
Hi @Camarine11 

There is font online compiler, kindly provided by @RoboDurden: https://pionierland.de/fonts/index.php

As stated here: https://github.com/lexus2k/lcdgfx/wiki/How-to-create-new-font-for-the-library#creating-new-font-using-GLCD-font-creator-fixed-width-fonts, you can use  LCD font generator from MikroC to generate new fonts for the library.

Please, use `new format` option of `fontgenerator.py` script to eliminate all empty vertical lines.

If anything doesn't work, let me know

Camarine11, 2021-08-10:
Hi lexus2k

Thank you for quick answer.

I tested as follows and had partial success:

Used Arduino IDE v1.8.13 set to Arduino Pro mini 3.3V 8Mhz and SH1106 128x64 screen with lcdgfx library v1.1.1 installed and USBasp programmer.
Used Python 3.6 with PIP 21.2.2,  freetype-py v2.2.0 and freetype.DLL v2.11.0 made for Windows.

1. Generated the large font with MikroC GLCD and tried running the python script as per wiki
 python fontgenerator.py --glcd glcdfont.c -f old > ssd1306font.h

I always get next error (this might be because I use Windows version of the Python):

Traceback (most recent call last):
  File "C:\Users\RMPC\Documents\Arduino\libraries\lcdgfx\to
, line 176, in <module>
    source = glcdsource.GLCDSource(fname, fsize, "utf8")
  File "C:\Users\RMPC\Documents\Arduino\libraries\lcdgfx\to
e.py", line 79, in __init__
    bitmap[y].append( (bytes[row] >> bit) & 0x01 )
TypeError: list indices must be integers or slices, not float.

This is the content of the glcdfont.c file from GLCD I used:

```

//WARNING: This Font Require X-GLCD Lib.
//         You can not use it with MikroE GLCD Lib.

//Font Generated by MikroElektronika GLCD Font Creator 1.2.0.0
//MikroElektronika 2011 
//http://www.mikroe.com 

//GLCD FontName : Impact33x55
//GLCD FontSize : 33 x 55

const unsigned short Impact33x55[] = {
        0x21, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0xFF, 0xFF, 0xFF, 0x0F, 0x00, 0x80, 0xFF, 0xFF, 0xFF, 0xFF, 0x7F, 0x00, 0xC0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x01, 0xF0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x07, 0xF8, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x0F, 0xF8, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x0F, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x7F, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x7F, 0xFF, 0x01, 0x00, 0x00, 0x00, 0xC0, 0x7F, 0xFF, 0x00, 0x00, 0x00, 0x00, 0x80, 0x7F, 0xFF, 0x00, 0x00, 0x00, 0x00, 0x80, 0x7F, 0xFF, 0x00, 0x00, 0x00, 0x00, 0x80, 0x7F, 0xFF, 0x01, 0x00, 0x00, 0x00, 0xC0, 0x7F, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x7F, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x7F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x7F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0xF8, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x0F, 0xF0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x0F, 0xE0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x07, 0xC0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x01, 0x00, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x00, 0x00, 0xF8, 0xFF, 0xFF, 0xFF, 0x1F, 0x00,  // Code for char 0
        0x17, 0x00, 0x7E, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7E, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7E, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x7F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0xFF, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC0, 0xFF, 0x01, 0x00, 0x00, 0x00, 0x00, 0xC0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xE0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xF0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xF0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xF8, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,  // Code for char 1
        0x1F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFC, 0x07, 0x00, 0x00, 0x80, 0x3F, 0x80, 0xFF, 0x07, 0x00, 0x00, 0xE0, 0x3F, 0xE0, 0xFF, 0x07, 0x00, 0x00, 0xF8, 0x3F, 0xF0, 0xFF, 0x07, 0x00, 0x00, 0xFE, 0x3F, 0xF8, 0xFF, 0x07, 0x00, 0x80, 0xFF, 0x3F, 0xFC, 0xFF, 0x07, 0x00, 0xE0, 0xFF, 0x3F, 0xFC, 0xFF, 0x07, 0x00, 0xF8, 0xFF, 0x3F, 0xFE, 0xFF, 0x07, 0x00, 0xFE, 0xFF, 0x3F, 0xFE, 0xFF, 0x07, 0x00, 0xFF, 0xFF, 0x3F, 0xFF, 0xFF, 0x07, 0xC0, 0xFF, 0xFF, 0x3F, 0xFF, 0xFF, 0x07, 0xF0, 0xFF, 0xFF, 0x3F, 0xFF, 0x03, 0x00, 0xFC, 0xFF, 0xFF, 0x3F, 0xFF, 0x00, 0x00, 0xFF, 0xFF, 0xFF, 0x3F, 0xFF, 0x00, 0xC0, 0xFF, 0xFF, 0xFF, 0x3F, 0xFF, 0x00, 0xF8, 0xFF, 0xFF, 0xFF, 0x3F, 0xFF, 0x01, 0xFF, 0xFF, 0xFF, 0xDF, 0x3F, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xCF, 0x3F, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xC3, 0x3F, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xC0, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0x3F, 0xC0, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0x1F, 0xC0, 0x3F, 0xFE, 0xFF, 0xFF, 0xFF, 0x07, 0xC0, 0x3F, 0xFC, 0xFF, 0xFF, 0xFF, 0x01, 0xC0, 0x3F, 0xF8, 0xFF, 0xFF, 0x7F, 0x00, 0xC0, 0x3F, 0xF8, 0xFF, 0xFF, 0x1F, 0x00, 0xC0, 0x3F, 0xF0, 0xFF, 0xFF, 0x07, 0x00, 0xC0, 0x3F, 0xC0, 0xFF, 0xFF, 0x01, 0x00, 0xC0, 0x3F, 0x80, 0xFF, 0x7F, 0x00, 0x00, 0xC0, 0x3F, 0x00, 0xFC, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x [... cut ...]
Testing I found I get garbles data on SH1105 screen even when using built-in variable size fonts type 0x02: I used free_calibri11x12 from fonts.c file and still getting garbled data on the screen.
I get garbled data when using any font created by fontgenerator.py  using the -new format command.

Had similar result when used the font file generated by online tool from https://www.pionierland.de/fonts.
Since I also get garbled data with built-in font type 0x02 I think the online tool generates the font file code correctly.

Cheers

lexus2k, 2021-08-11:
Thank you for reporting the issue.
if you have any screenshots to attach, that would be very helpful.

closed 2026-04-03 by lexus2k