lexus2k/lcdgfx #71: SSD1351 OLED C issues

ITstreet1, 2021-11-16:
I have an OLED C from MikroE. This is an SSD1351 96X96 SPI display.

With these examples here, I get flipped text, or whatever. Besides library is written for 128x128 so I get an offset on the display.
Is there any way of tweaking this library to make it work,

lexus2k, 2021-11-18:
Hi, can you send me screenshots? 
Actually, it is possible to support 96x96 display, if you have some time to do the tests for me.
Let me know

ITstreet1, 2021-11-18:
I have time, of course. 
Screenshots here?

ITstreet1, 2021-11-18:
`#include "lcdgfx.h"
#include "lcdgfx_gui.h"

DisplaySSD1351_128x128x16_SPI display(34,{-1, 87, 7, 0,-1,-1});
uint8_t rotation = 0;
void setup() {
  pinMode(27, OUTPUT); //EN
  //pinMode(55, OUTPUT); //R/W
  digitalWrite(27, HIGH);
  //digitalWrite(55, LOW);
    display.begin();
    display.setFixedFont(ssd1306xled_font6x8);

    display.fill( 0x0000 );
   // menu.show( display );
    display.setFixedFont(ssd1306xled_font6x8);
    display.clear();
    display.setColor(RGB_COLOR16(255,255,0));
    display.printFixed(0,  8, "Normal text", STYLE_NORMAL);
    display.setColor(RGB_COLOR16(0,255,0));
    display.printFixed(0, 16, "bold text?", STYLE_BOLD);
    display.setColor(RGB_COLOR16(0,255,255));
    display.printFixed(0, 24, "Italic text?", STYLE_ITALIC);
    display.setColor(RGB_COLOR16(255,255,255));
    display.invertColors();
    display.printFixed(0, 32, "Inverted bold?", STYLE_BOLD);
    display.invertColors();
    lcd_delay(3000);
}

// the loop function runs over and over again forever
void loop() {
    
}
`
This is just a strip-off of your example. And this is what I get

[image]
:

As you can see RGB pattern is not good. There is an offset on the right and flip.

lexus2k, 2021-11-19:
Hi

The `DisplaySSD1351_128x128x16_SPI` has `getInterface().setRotation()`, `getInterface().setStartBlock()` and `getInterface().setRgbMode()` methods. Can you play with them? Maybe there is something useful that will fix some of your issues.

Also, I found some link for your hardware: https://www.mikroe.com/oled-c-click. Is this the one you have?

ITstreet1, 2021-11-19:
Yup, that is the one. I forgot to put a link. Sry.

I will try and let you know. What about a different resolution? Can this be tweaked somehow?

ITstreet1, 2021-11-19:
Where can I find parameters for those methods? In an example, there is something like:
display.getInterface().setRotation((++rotation) & 0x03);
What is that HEX 0x03 for?

And besides, I need a flip, not rotation.

For that setStartBlock, I found nothing. There are so many files in this lib.

Is there a file with all available functions?

lexus2k, 2021-11-22:
There is the link the documentation on the first github page.

Some links for you are below:
https://codedocs.xyz/lexus2k/lcdgfx/class_interface_s_s_d1351.html#afa34ecadaed45d4cbbfa1be27354de9a
https://codedocs.xyz/lexus2k/lcdgfx/class_interface_s_s_d1351.html

ITstreet1, 2021-11-22:
LOL, I miss that :)

setRgbeMode() fixed one issue.

Before I get into the offset problem, I should fix that flip. With setRotation() I can only rotate the display, but can not flip it back.
Any help with this?

lexus2k, 2021-11-22:
Since I don't have official datasheet for your display to understand how OLED display is connected to the display controller (that's not MCU), I can advise you to play with setRotation() implementation for ssd1351 display.
Let me know how it goes.

```.cpp
    uint8_t ram_mode;
    if ( (rotation ^ m_rotation) & 0x01 )
    {
        m_base.swapDimensions();
    }
    m_rotation = (rotation & 0x03);
    this->start();
    setDataMode(0);
    this->send( 0xA0 );
    switch ( m_rotation )
    {
        // NORMAL FULL COLOR MODE
        case 0: // 0 degree CW
            ram_mode = 0b00110000; // Try change this to 0b00110010  or 0b00100000
            break;
        case 1: // 90 degree CW
            ram_mode = 0b00110011; // Try change this to 0b00110001 or 0b00100011
            break;
        case 2: // 180 degree CW
            ram_mode = 0B00100010; // Try change this to 0B00100000 or 0B00110010
            break;
        case 3: // 270 degree CW
        default:
            ram_mode = 0b00100001; // Try change this to 0b00100011 or 0b00110001
            break;
    }
    setDataMode(1); // According to datasheet all args must be passed in data mode
    this->send( ram_mode | m_rgbMode );
    this->stop();
```

> Will the resolution of this display cause additional problems?

Definitely, not.

ITstreet1, 2021-11-23:
Maybe this will do any help:
https://download.mikroe.com/documents/datasheets/psp27801-product-specification-a0.pdf

I will try above and let you know.

ITstreet1, 2021-11-23:
setRotation(0)

[image]

setRoration(1)

[image]

setRoration(2)

[image]

setRotation(3)

[image]

Sketch as follows:
`
#include "lcdgfx.h"
#include "lcdgfx_gui.h"
DisplaySSD1351_128x128x16_SPI display(34,{-1, 87, 7, 0,-1,-1});
void setup() {
  pinMode(27, OUTPUT); //EN
  //pinMode(55, OUTPUT); //R/W
  digitalWrite(27, HIGH);
  //digitalWrite(55, LOW);
    display.begin();
    display.fill( 0x0000 );
    display.getInterface().setRotation(0);
    display.setFixedFont(ssd1306xled_font6x8);
    display.getInterface().setRgbMode(0);
    display.clear();
    display.setColor(RGB_COLOR16(255,0,0));
    display.printFixed(0,  8, "Normal text", STYLE_NORMAL);
    display.setColor(RGB_COLOR16(0,255,0));
    display.printFixed(0, 16, "bold text?", STYLE_BOLD);
    display.setColor(RGB_COLOR16(0,0,255));
    display.printFixed(0, 24, "Italic text?", STYLE_ITALIC);
    display.setColor(RGB_COLOR16(255,255,255));
    display.invertColors();
    display.printFixed(0, 32, "Inverted bold?", STYLE_BOLD);
    display.invertColors();
    lcd_delay(3000);
}

// the loop function runs over and over again forever
void loop() {
    
}
`

That is the result.

ITstreet1, 2021-11-23:
``

ITstreet1, 2021-11-23:
As much I try these methods, it looks more and more that the image of this display is "running out" from the middle of the screen.

lexus2k, 2021-11-28:
> Maybe this will do any help: https://download.mikroe.com/documents/datasheets/psp27801-product-specification-a0.pdf

That's what exactly I need. Thank you

lexus2k, 2021-11-28:
I made initial commit for your display. Now there is `DisplaySSD1351_96x96x16_SPI()` constructor available.
However, you still need to do experiments with ram_mode values, I wrote above, and there is new function `display.getInterface().setOffset(x,y)`. Please, try it.

ITstreet1, 2021-11-30:
When I replaced:
DisplaySSD1351_128x128x16_SPI display(34,{-1, 87, 7, 0,-1,-1});
with this:
DisplaySSD1351_96x96x16_SPI display(34,{-1, 87, 7, 0,-1,-1});

From this

[image]
:

I get this:
[image]

Everything else is the same as above.
You can see that the red line of text has gone, and the bottom line has changed too. The white background becomes somewhat pink.

I didn't mention the connections. I use Flip&Click SAM3X and mikroBUS B. Here you can check all the data:
https://www.mikroe.com/flip-n-click-sam3x
And here are pin definitions:
https://github.com/MikroElektronika/Flip_n_click_Examples/blob/master/Breathalyzer/Breathalyzer/flip_click_defs.h

Maybe you can find some additional issues.

ITstreet1, 2021-11-30:
Ok, I played with the values. So far, so good.
With this sketch:
`#include "lcdgfx.h"
#include "lcdgfx_gui.h"
DisplaySSD1351_96x96x16_SPI display(34,{-1, 87, 7, 0,-1,-1});
//DisplaySSD1351_128x128x16_SPI display(34,{-1, 87, 7, 0,-1,-1});
void setup() {
  pinMode(27, OUTPUT); //EN
  //pinMode(55, OUTPUT); //R/W
  digitalWrite(27, HIGH);
  //digitalWrite(55, LOW);
    display.begin();
    display.fill( 0x0000 );
    display.getInterface().setRotation(0);
    display.getInterface().setOffset(16,16); //offset
    display.setFixedFont(ssd1306xled_font6x8);
    display.getInterface().setRgbMode(0);
    display.clear();
    display.setColor(RGB_COLOR16(255,0,0));
    display.printFixed(0,  8, "Normal text", STYLE_NORMAL);
    display.setColor(RGB_COLOR16(0,255,0));
    display.printFixed(0, 16, "bold text?", STYLE_BOLD);
    display.setColor(RGB_COLOR16(0,0,255));
    display.printFixed(0, 24, "Italic text?", STYLE_ITALIC);
    display.setColor(RGB_COLOR16(255,255,255));
    display.invertColors();
    display.printFixed(0, 32, "Inverted bold?", STYLE_BOLD);
    display.invertColors();
    lcd_delay(3000);
}

// the loop function runs over and over again forever
void loop() {
    
}`

I get this:

[image]

It is well-positioned. Except it is flipped. The colors seem good, but this white background is somewhat pink. Now just to fix this flip effect and this white->pink (255,255,255)

lexus2k, 2021-12-01:
Did you try to change ram_mode to new values?

```.cpp
    switch ( m_rotation )
    {
        // NORMAL FULL COLOR MODE
        case 0: // 0 degree CW
            ram_mode = 0b00110000; // Try change this to 0b00110010  or 0b00100000
            break;
        case 1: // 90 degree CW
            ram_mode = 0b00110011; // Try change this to 0b00110001 or 0b00100011
            break;
        case 2: // 180 degree CW
            ram_mode = 0B00100010; // Try change this to 0B00100000 or 0B00110010
            break;
        case 3: // 270 degree CW
        default:
            ram_mode = 0b00100001; // Try change this to 0b00100011 or 0b00110001
            break;
    }
```

ITstreet1, 2021-12-01:
Ok, there is some progress...

**Rotation 0**

setRotation(0);
setOffset(16,16);
ram_mode = 0b00110010;

Everything is fine. That pink-white has lost. Now there is white as it should be.

[image]

**Rotation 1**

setRotation(1);
setOffset(16,16);
ram_mode = 0b00110001;

Here is everything ok, too.

[image]

**Rotation 2**

setRotation(2);
setOffset(16,16);
ram_mode = 0B00100000;

This does not work as it should.

[image]

setRotation(2);
setOffset(16,32);
ram_mode = 0B00100000;

[image]

setRotation(2);
setOffset(16,-6);
ram_mode = 0B00100000;

[image]

Only ram_mode = 0B00100000; fixes the flip. Just couldn't position the offset. Values for y above the 32 and below the -6 make no sense. Anything between doesn't fix it. You can see it in the pictures.

Value for x is ok, it positions the text correctly, but there is a snow as you can see. 

For rotation(3), it looks almost the same as for rotation(2).

lexus2k, 2021-12-02:
First, please try to set offset after display initialization, and then you will be able to rotate the display.
The library must automatically recalculate required offset for COL/SEG lines.

```.cpp
    display.begin();
    display.getInterface().setOffset(16,32); // I still believe there should be 16 and 32. Check the datasheet above
    display.getInterface().setRotation( NUMBER_FROM_0_TO_3 );
```

ITstreet1, 2021-12-02:
Ok
With this:

```
#include "lcdgfx.h"
#include "lcdgfx_gui.h"
DisplaySSD1351_96x96x16_SPI display(34,{-1, 87, 7, 0,-1,-1});
void setup() {
  pinMode(27, OUTPUT); //EN
  //pinMode(55, OUTPUT); //R/W
  digitalWrite(27, HIGH);
  //digitalWrite(55, LOW);
    display.begin();
    display.getInterface().setOffset(16,16); //offset 16,16 for rotation 0 and 1
    display.getInterface().setRotation(0);
    display.fill( 0x0000 );
    display.setFixedFont(ssd1306xled_font6x8);
    display.getInterface().setRgbMode(0);
    display.clear();
    display.setColor(RGB_COLOR16(255,0,0));
    display.printFixed(0,  8, "Normal text", STYLE_NORMAL);
    display.setColor(RGB_COLOR16(0,255,0));
    display.printFixed(0, 16, "bold text?", STYLE_BOLD);
    display.setColor(RGB_COLOR16(0,0,255));
    display.printFixed(0, 24, "Italic text?", STYLE_ITALIC);
    display.setColor(RGB_COLOR16(255,255,255));
    display.invertColors();
    display.printFixed(0, 32, "Inverted bold?", STYLE_BOLD);
    display.invertColors();
    lcd_delay(3000);
}
void loop() {
}
```
I get the right position, colors, etc for Rotation 0 and 1. I posted the ram_mode above for those two. They are ok.

Rotations 2 and 3 can not be fixed. Flip is ok, RGB is ok, but the position is not. Just as I said above, values for y can go from -6 to 32 but I can not set it as it should be.

[image]

One weird thing for rotation 3:

setRotation(3);
setOffset(16,32);
ram_mode = 0b00100011;

[image]

setRotation(3);
setOffset(16,48);
ram_mode = 0b00100011;

[image]

As you can see, y=32 is "not enough" but if I go for y=48 I can get a text on the screen but there is snow. And x is not good. I will have to play a little bit with it.

lexus2k, 2021-12-12:
Hi,

I checked ssd1351 datasheet (https://www.newhavendisplay.com/app_notes/SSD1351.pdf), verified the code, and still didn't find the root case for the white noise on the last picture by you.

ITstreet1, 2021-12-13:
What do you suggest I should try?

ITstreet1, 2021-12-14:
> I made initial commit for your display. Now there is `DisplaySSD1351_96x96x16_SPI()` constructor available.
> However, you still need to do experiments with ram_mode values, I wrote above, and there is new function `display.getInterface().setOffset(x,y)`. Please, try it.

Little OFF-TOPIC.

Can you please point me to these constructor parameters? There are some -1 values. I couldn't find out what they stand for. Say I want to connect this display to an SPI2, or maybe SPI3. What to change to point to other SPI than the default? So I would like to point SPI to other MOSI, MISO, and SCK. Is it posible?