lexus2k/lcdgfx #73: please add scl/sda comment for DisplaySSD1306_128x64_I2C display(-1);

RoboDurden, 2021-11-22:
Please change

`//DisplaySSD1306_128x64_I2C display(-1); // This line is suitable for most platforms by default`

into 

`//DisplaySSD1306_128x64_I2C display(-1); // or (-1,{busId, addr, scl, sda, frequency}); https://codedocs.xyz/lexus2k/lcdgfx/struct_s_platform_i2c_config.html
`

It is so frustrating when you first have to search 30 minutes all your code to get the demo running. 
ESP32 here with custom sda and scl pins.
thanks.

lexus2k, 2021-11-22:
Hi, I'm ok with the comment except the link.
I prefer not to use direct links to the documentation, as the URL is generated automatically by the side tool.

RoboDurden, 2021-11-23:
:-)
Greetings from Germany.

closed 2021-11-23 by RoboDurden