lexus2k/lcdgfx #74: how to clearRect or setColor on SSD1306 ??

RoboDurden, 2021-11-24:
Sorry for another simple question :-/
Have found nothing in your nice https://codedocs.xyz/lexus2k/lcdgfx/index.html

I only want to clear a small region like a `clearRect(0,0,32,16);` to prevent the flickering of `display.clear();`.
Or is there a way to set the color to black like `display.setColor(RGB_COLOR8(0,0,0));` ?

My workaround: `display.printFixed(x, y, "    ");`

Thanks again :-)

lexus2k, 2021-11-25:
Hi 

> I only want to clear a small region like a clearRect(0,0,32,16); to prevent the flickering of display.clear();.
> Or is there a way to set the color to black like display.setColor(RGB_COLOR8(0,0,0)); ?

The way for you is to use `fillRect()` and `setColor()` methods.

Best regards

RoboDurden, 2021-11-25:
Thank you, this indeed works:
```
      display.setColor(RGB_COLOR8(0,0,0));
      display.fillRect(x,y,x+32,y+16);

```
I had only tested this with no success:
```
      display.setColor(RGB_COLOR8(0,0,0));
      display.printFixed(x, y, s, STYLE_BOLD);
```
This however  has an effect:
```
      display.setColor(RGB_COLOR8(0,0,0));
      display.invertColors();
      display.printFixed(x, y, s, STYLE_BOLD);
```
That setColor(0) makes invertColors having no effect.

setColor(255) enables it again:
 ```
   display.setColor(RGB_COLOR8(255,255,255));
   display.invertColors();    
   display.printFixed(x, y, s, STYLE_BOLD);
```
So what confuses me is that this has no effect:
```
    display.setColor(RGB_COLOR8(0,0,0));
   display.printFixed(x, y, s, STYLE_BOLD);
```
So for `printFixed`, the `setColor` function is rather a `setBackgroundColor`.
Whereas for `fillRect` the `setColor` function is setting the paint color.

It is still nice because printing text will overwrite the area anyway (which i had not been aware of and therefore wanted a clearRect). So no need to set background and forground color but one call to `invertColor` will do it.

A bit confusing but i have no idea for you how to do it better :-)