lexus2k/lcdgfx #76: How to print a variable?

raphael303, 2021-12-23:
Ask any question, you have, regarding the library
How can I print a variable?

I tried: 
`void loop()
{
    display.printFixed(0,  8, "Counter", STYLE_NORMAL);
    display.printFixed(0,  16, counter, STYLE_NORMAL);
    counter = counter + 1;
    delay(1000);
}`
Which display gibberish instead of the counter value. Also tried a "float" variable, which doesn't compile at all. So how do I display variables of any kind? I looked at most of the demos, but didn't find an example for this.

Thank you.

lexus2k, 2022-02-21:
Use `display.print(10)` or `display.print(2.2f)`.
If that doesn't work for, let me know.

closed 2022-02-21 by lexus2k

TurboMarian, 2022-07-10:
Can we have example of this? When I use :
display.print (80, 16, (OILP, 0), STYLE_BOLD);
have error:
no matching function for call to 'DisplaySSD1351_128x128x16_SPI::print(int, int, int, EFontStyle)'

Thank you.