lexus2k/lcdgfx #77: The drawLine function is missing pixels on 4-bit grayscale SSD1327 display.

kfud, 2022-01-02:
**Describe the bug**
Hi, I noticed when drawing lines with the display.drawLine() function that it seems to draw broken lines. Most of the time it's missing every second pixel.

**To Reproduce**
Steps to reproduce the behavior:

Drawing lines vertically or horizontally with the example code below demonstrates the problem:

static void gradientDemo() {
     int color = 0;
    for (uint8_t y = 0; y < display.height(); y++) {
        display.setColor(GRAY_COLOR4(y * 2));
        display.drawLine(y, 0, y, display.height());
    }
    delay(3000);
}

Using drawH/VLine works correctly and draws complete lines.

**Expected behavior**
Complete lines should be drawn.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**
 - library version: 1.1.1
 - SSD1327 128x128 4-bit OLED
 - OS : Linux
 - Platform: Arduino (Every) [Atmega4809]
 - IDE: Arduino

lexus2k, 2022-01-09:
Hi

Let me clarify the behavior. There are 2 sets of the drawing functions available in the library: direct draw, and canvas draw.
Direct draw functions do not use any RAM, they send all pixel data directly to the display controller. There is no way for the library to read the GDRAM back from the controller, since neither SPI or I2C interfaces of the display allow to do that. DrawLine function is generic function for all supported LCD displays. and it draws the lines pixel by pixel. Since each byte encodes 2 pixels at once, the new pixel overwrite old pixels data. 
As for drawVLine, drawHLine - they are specific for the particular display. That's why strange things happen.

[image]

If you have an idea how to fix the issue, let me know.
Maybe I will find the solution.

lexus2k, 2022-01-09:
Hi, I committed some fix for 4-bit displays. Let me know how it works for you.
It doesn't fix all the issues described above, but it makes some improvements.

kfud, 2022-01-10:
Hi! Thanks for looking into it! I just want to say, I've tried a lot of GUI libraries, and this is the only one that works well will gray-scale displays! Although I did buy a gray-scale display by accident, I'm happy there's a library that allows me to take full advantage of it! 

I'll try out the fix this week, I'm busy setting up my new 3d-printer to print the case for this project, so I'll keep you informed.

lexus2k, 2022-01-20:
Hi

now ssd1327 demo should look much better:

[image]

Let me know when you check the update library.

kfud, 2022-01-21:
Hi, thanks again for looking at this, super busy at the moment, hope I can try it out this weekend!

kfud, 2022-01-29:
Hi, I tried the fix, works great thanks!

closed 2022-01-29 by kfud